linrunner/TLP #798: tlp recalibrate BAT0 freezes laptop

danielktdoranie, 2025-05-05:
Hi so both

`tlp recalibrate BAT0`

and 

`tlp discharge`

cause my Lenovo ThinkPad Yoga S1 20c0 to freeze up.

`tlp fullcharge` does NOT cause my laptop to freeze.

Running Debian 12 completely up-to-date

This happens on both the original Sony made 45N1705 and the NEW SMP made 45N1705 batteries. 

Old Sony only held a 78% capacity so I bought a brand new SMP battery off Amazon and it only has a 71% capacity! So I wanted to recalibrate both using TLP.

danielktdoranie, 2025-05-05:
[](https://picsur.org/i/dbf4c4ca-713e-4b75-afc6-1fff0730dd1c.jpg)

So as soon as I enter the command `tlp discharge` this is what happens. Just freezes the whole computer, can’t move the mouse.

linrunner, 2025-05-07:
Hi, I classify this as a hardware or firmware problem.

The embedded controller (EC) controls the entire charging and discharging behavior as well as switching on and off the Thinkpad. In the FAQ you'll find tips on how to get a bitching EC to behave -> https://linrunner.de/tlp/faq/battery.html#various-thinkpad-battery-malfunctions

If that doesn't help, you should seek advice in a hardware-oriented forum, because the topic is out of scope here.

By the way, all the outputs required in the issue template are missing, without which there is no support here anyway.