linrunner/TLP #802: USB-C  docking recognizing

Condorello, 2025-06-11:
Hi, I'm trying to trigger the TLP network wizard on dock/undock events. After some testing, I discovered that TLP recognizes a docked notebook only if a docked attribute appears under /sys/devices/platform/dock*.
However, newer USB-C docking stations do not create this device and are instead recognized simply as USB hubs.
It would be very useful if TLP allowed custom configuration to detect docking status based on the presence of a specific vendor/device ID, so that the system could be considered "docked" when such a device is found.
Thanks in advance for your attention and for your great work on TLP.

linrunner, 2025-06-15:
Hi. Since you are using RDW, I assume that a LAN cable is connected to the dock. Why don't you use 

    DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi wwan"
    DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi wwan"

then? I personally prefer and use this.

The feature you suggest is complicated to configure, as you first have to find out which USB devices actually belong to the dock.

linrunner, 2025-06-16:
I see. You should have mentioned that. Let's see what happens:

1. Enable the trace by adding `TLP_DEBUG="lock nm rf run"`  to your configuration
2. Shutdown the system
3. Start the system
4. Show the full output of

```
sudo tlp-stat
journalctl -g 'tlp|NetworkManager' -b 0
```

Condorello, 2025-06-16:
Thanks for you attention, here the resutls:

`    ~  sudo tlp-stat                                                    ✔ 
Posizionare il dito sul lettore di impronte
--- TLP 1.8.0 --------------------------------------------

+++ Configured Settings:
defaults.conf L0006: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
defaults.conf L0008: TLP_MSG_COLORS="91 93 1 92"
defaults.conf L0009: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0010: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0011: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0012: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0013: MAX_LOST_WORK_SECS_ON_BAT="60"
/etc/tlp.conf L0147: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
/etc/tlp.conf L0148: CPU_ENERGY_PERF_POLICY_ON_BAT="power"
defaults.conf L0016: NMI_WATCHDOG="0"
defaults.conf L0017: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0018: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0019: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0020: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0021: DISK_IOSCHED="keep keep"
defaults.conf L0022: SATA_LINKPWR_ON_AC="med_power_with_dipm"
/etc/tlp.conf L0274: SATA_LINKPWR_ON_BAT="min_power"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0025: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0026: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0027: PCIE_ASPM_ON_AC="default"
/etc/tlp.conf L0393: PCIE_ASPM_ON_BAT="powersupersave"
/etc/tlp.conf L0332: RADEON_DPM_PERF_LEVEL_ON_AC=""
/etc/tlp.conf L0333: RADEON_DPM_PERF_LEVEL_ON_BAT=""
/etc/tlp.conf L0347: RADEON_POWER_PROFILE_ON_AC="" #! Parameter is deprecated: feature will be removed with the next release
/etc/tlp.conf L0348: RADEON_POWER_PROFILE_ON_BAT="" #! Parameter is deprecated: feature will be removed with the next release
/etc/tlp.conf L0358: AMDGPU_ABM_LEVEL_ON_AC=""
/etc/tlp.conf L0359: AMDGPU_ABM_LEVEL_ON_BAT=""
defaults.conf L0035: WIFI_PWR_ON_AC="off"
defaults.conf L0036: WIFI_PWR_ON_BAT="on"
defaults.conf L0037: WOL_DISABLE="Y"
defaults.conf L0038: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0039: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0040: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0041: BAY_POWEROFF_ON_AC="0"
defaults.conf L0042: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0043: BAY_DEVICE="sr0"
defaults.conf L0044: RUNTIME_PM_ON_AC="on"
defaults.conf L0045: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0046: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon xhci_hcd"
defaults.conf L0047: USB_AUTOSUSPEND="1"
defaults.conf L0048: USB_EXCLUDE_AUDIO="1"
defaults.conf L0049: USB_EXCLUDE_BTUSB="0"
defaults.conf L0050: USB_EXCLUDE_PHONE="0"
defaults.conf L0051: USB_EXCLUDE_PRINTER="1"
defaults.conf L0052: USB_EXCLUDE_WWAN="0"
/etc/tlp.conf L0482: RESTORE_DEVICE_STATE_ON_STARTUP=""
/etc/tlp.conf L0552: RESTORE_THRESHOLDS_ON_BAT=""
defaults.conf L0055: NATACPI_ENABLE="1"
defaults.conf L0056: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0031: TLP_DEBUG="lock nm rf run"
/etc/tlp.conf L0117: CPU_SCALING_GOVERNOR_ON_AC="performance"
/etc/tlp.conf L0118: CPU_SCALING_GOVERNOR_ON_BAT="powersave"
/etc/tlp.conf L0168: CPU_BOOST_ON_AC="1"
/etc/tlp.conf L0169: CPU_BOOST_ON_BAT="0"
/etc/tlp.conf L0179: CPU_HWP_DYN_BOOST_ON_AC="1"
/etc/tlp.conf L0180: CPU_HWP_DYN_BOOST_ON_BAT="1"
/etc/tlp.conf L0196: PLATFORM_PROFILE_ON_AC="balanced"
/etc/tlp.conf L0197: PLATFORM_PROFILE_ON_BAT="low-power"
/etc/tlp.conf L0563: DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi wwan"
/etc/tlp.conf L0569: DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi"

[... 193 lines cut ...]
giu 16 18:04:21 andrea-notebook NetworkManager[817]: <info>  [1750089861.1604] Loaded device plugin: NMWwanFactory (/usr/lib/Netw>
giu 16 18:04:21 andrea-notebook NetworkManager[817]: <info>  [1750089861.1614] Loaded device plugin: NMOvsFactory (/usr/lib/Netwo>
giu 16 18:04:21 andrea-notebook NetworkManager[817]: <info>  [1750089861.1625] Loaded device plugin: NMWifiFactory (/usr/lib/Netw>
giu 16 18:04:21 andrea-notebook NetworkManager[817]: <info>  [1750089861.1855] Loaded device plugin: NMTeamFactory (/usr/lib/Netw>
giu 16 18:04:21 andrea-notebook NetworkManager[817]: <info>  [1750089861.1899] manager: (lo): new Loopback device (/org/freedeskt>
giu 16 18:04:21 andrea-notebook NetworkManager[817]: <info>  [1750089861.1921] manager: (wlp0s20f3): new 802.11 Wi-Fi device (/or>
giu 16 18:04:21 andrea-notebook NetworkManager[817]: <info>  [1750089861.1934] bus-manager: acquired D-Bus service "org.freedeskt>
giu 16 18:04:21 andrea-notebook tlp[876]: lock_tlp_nb().failed
giu 16 18:04:21 andrea-notebook tlp[665]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpiOwito -> /run/tlp/run.conf
giu 16 18:04:21 andrea-notebook tlp[665]: unlock_tlp()
giu 16 18:04:21 andrea-notebook tlp[1077]: lock_tlp_nb().success
giu 16 18:04:21 andrea-notebook tlp[1077]: unlock_tlp()
giu 16 18:04:21 andrea-notebook NetworkManager[817]: <info>  [1750089861.4335] manager: (p2p-dev-wlp0s20f3): new 802.11 Wi-Fi P2P>
giu 16 18:04:22 andrea-notebook NetworkManager[817]: <info>  [1750089862.8970] manager: (eth0): new Ethernet device (/org/freedes>
giu 16 18:04:25 andrea-notebook NetworkManager[817]: <info>  [1750089865.0805] manager: NetworkManager state is now CONNECTING
giu 16 18:04:25 andrea-notebook NetworkManager[817]: <info>  [1750089865.2787] manager: NetworkManager state is now CONNECTED_SITE
giu 16 18:04:25 andrea-notebook NetworkManager[817]: <info>  [1750089865.3514] manager: NetworkManager state is now CONNECTED_GLO>
giu 16 18:04:25 andrea-notebook tlp[1806]: lock_tlp().success
[... cut ...] ea-notebook tlp[1806]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpo149LO -> /run/tlp/run.conf
giu 16 18:04:25 andrea-notebook tlp[1806]: unlock_tlp()
giu 16 18:04:26 andrea-notebook tlp[1981]: lock_tlp_nb().success
giu 16 18:04:26 andrea-notebook tlp[1981]: unlock_tlp()
giu 16 18:04:31 andrea-notebook systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
giu 16 18:04:50 andrea-notebook sudo[3208]:   andrea : TTY=pts/0 ; PWD=/home/andrea ; USER=root ; COMMAND=/usr/bin/tlp-stat
`

linrunner, 2025-06-16:
This seems to be your the dock's USB ethernet device:

> Bus 003 Device 009 ID **0b95:7720** control = auto, autosuspend_delay_ms = 2000 -- ASIX Electronics Corp. AX88772 (asix)

A shot in the dark wouodl be to exclude it from autosuspend:

USB_DENYLIST="0b95:7720"

By the way, I also have an X1C9 here, and it works fine with a Realtek RTL8153 USB gigabit ethernet adapter.

linrunner, 2025-06-19:
> Just out of curiosity: why do you consider it difficult to implement dock/undock detection with USB docking stations? The dock itself is recognized as a device

With "complicated" I didn't mean the implementation, I meant that the configuration is tedious for the user. Users would have to identify an USB device belonging the dock, which isn't always as straightforward as in your case. For example ThinkPad CS18 docks (which are USB-C internally) can be recognized by an undistinguished audio device only.

Instead of implementing a workaround for non-functioning `DEVICES_TO_DISABLE/ENABLE_ON_LAN_...`, I would rather know why it doesn't work for you.

----

I need some more trace information and would also like to know if the LAN is actually connected after system startup.

Please change the config to `TLP_DEBUG="lock nm rf run usb"` and reboot again. Then show

    journalctl -g 'tlp|NetworkManager' -b 0
    sudo tlp-stat -u 
    nmcli con show --active

linrunner, 2025-06-25:
@Condorello Any news on this?

linrunner, 2025-07-01:
Oh dear, I should have asked that first; or looked at your output, because it's there:

> RDW state = not installed

Thanks for the feedback.

closed 2025-07-01 by linrunner

linrunner, 2025-07-01:
tlp.conf and docs updated.

linrunner, 2025-07-16:
For good measure I also added a warning` tlp-stat -s`