linrunner/TLP #803: Battery Care for Tuxedo/Clevo Laptops

almndtr33, 2025-06-14:
I own a Tuxedo laptop and would like to control the charge levels using TLP. There is a [driver](https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers) which provides the relevant files in `/sys/class/power_supply`, so I started adapting the template: https://github.com/almndtr33/TLP/blob/main/bat.d/70-tuxedo

As far as I can tell it works. Should I create a pull request?

There are still the following issues:
- I am not sure about the correct implementation in some of the places marked with TEMPLATE, mainly how far the values found on my system would generalize to all Tuxedo hardware.
- While TLP wrote the charge thresholds correctly using the file linked above, I had to run ´echo "Custom" > /sys/class/power_supply/BAT0/charge_type` to make my laptop respect the configured charge thresholds.

## Alternatives

There is a [feature request in tuxedo-rs](https://github.com/AaronErhardt/tuxedo-rs/issues/80) which would add the support for charge thresholds. However, it is not yet fully implemented.

linrunner, 2025-06-16:
Hi, 

my first obstacle is the driver zoo that Tuxedo provides. Charge thresholds are implemented in [clevo_keyboard.h](https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers/-/blob/main/src/clevo_keyboard.h?ref_type=heads#L775). Mind you **.h** not **.c**. In its current form, its chances of making it into Linus' upstream kernel are pretty slim imho.

TLP usually only provides plugins that are based on upstream kernel drivers - or at least those that are already submitted upstream (or about to be).

But let's assume for the moment that the prospective user has installed the driver and let me take a look at your plugin:

1. It is essential that the plugin checks the value ranges for start and stop/end threshold. This happens in [batdrv_write_thresholds ()](https://github.com/almndtr33/TLP/blob/0cb31c05713991def0901c5d596abac2a740c54a/bat.d/70-tuxedo#L259C1-L259C29) and you have to customize it. The [kernel driver](https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers/-/blob/main/src/clevo_keyboard.h?ref_type=heads#L775) considers the range 1..100 as valid for both.
2. [batdrv_show_battery_data ()](https://github.com/almndtr33/TLP/blob/0cb31c05713991def0901c5d596abac2a740c54a/bat.d/70-tuxedo#L584) also shows the ranges and has to be customized.
3. The hardware defaults for the threshold values must be customized (if they have never been configured before). These are defined in [batdrv_init ()](https://github.com/almndtr33/TLP/blob/0cb31c05713991def0901c5d596abac2a740c54a/bat.d/70-tuxedo#L87):   `_bt_def_start`,   `_bt_def_stop`.
4. `charge_type = Custom`: looks familiar to me, it is implemented in the **65-dell** plugin. Why don't you use that as a template?

I would love to see the next iteration of your plugin. Please show the output generated by it with:

    tlp-stat -s -b

Before you do all that, show me the output of (**TLP 1.8** required):

    tlp-stat --pev

linrunner, 2025-06-19:
> Running `tlp-stat --pev` appears to log events, however there are no events in "normal" operation.

I'm sorry, my bad. What I meant was

     tlp-stat --psup

almndtr33, 2025-06-19:
No problem!

```
# tlp-stat --psup
--- TLP 1.9.0-alpha.0 --------------------------------------------

+++ Power supply diagnostic
/sys/class/power_supply/AC/type: Mains
/sys/class/power_supply/AC/online: 1
/sys/class/power_supply/BAT0/type: Battery
/sys/class/power_supply/BAT0/present: 1
/sys/class/power_supply/BAT0/capacity: 100
/sys/class/power_supply/BAT0/charge_full_design: 9600000
/sys/class/power_supply/BAT0/charge_full: 8641000
/sys/class/power_supply/BAT0/charge_now: 8641000
/sys/class/power_supply/BAT0/current_now: 0
/sys/class/power_supply/BAT0/voltage_now: 8454000
/sys/class/power_supply/BAT0/charge_control_start_threshold: 95
/sys/class/power_supply/BAT0/charge_control_end_threshold: 100
/sys/class/power_supply/BAT0/status: Not charging
/sys/class/power_supply/BAT0/device/path: \_SB_.BAT0
/sys/class/power_supply/ucsi-source-psy-USBC000:001/type: USB
/sys/class/power_supply/ucsi-source-psy-USBC000:001/usb_type: C [PD] PD_PPS
/sys/class/power_supply/ucsi-source-psy-USBC000:001/online: 0
/sys/class/power_supply/ucsi-source-psy-USBC000:001/current_now: 3000000
/sys/class/power_supply/ucsi-source-psy-USBC000:001/voltage_max: 0
/sys/class/power_supply/ucsi-source-psy-USBC000:001/voltage_min: 0
/sys/class/power_supply/ucsi-source-psy-USBC000:001/voltage_now: 0

+++ udev diagnostic
Reading rules file: /nix/store/03x61jslfaz5f0599fr2k314cf06q071-udev-rules/85-tlp.rules
  RUN{program} : /nix/store/0pnmmi2fgg33yn68cmdc9c20423h37vf-tlp-1.8.0/sbin/tlp auto
OK.
```

almndtr33, 2025-07-06:
The remaining changes are implemented and I created a pull request. I will have to look into how to run the tests when the time comes, but it should not be a problem.

<details>
  <summary>The requested <tt>grep</tt> output</summary>
  
```
$ grep . /sys/class/power_supply/BAT0/*
/sys/class/power_supply/BAT0/alarm:0
/sys/class/power_supply/BAT0/capacity:62
/sys/class/power_supply/BAT0/capacity_level:Normal
/sys/class/power_supply/BAT0/charge_control_end_available_thresholds:60 70 80 90 100
/sys/class/power_supply/BAT0/charge_control_end_threshold:70
/sys/class/power_supply/BAT0/charge_control_start_available_thresholds:40 50 60 70 80 95
/sys/class/power_supply/BAT0/charge_control_start_threshold:60
/sys/class/power_supply/BAT0/charge_full:8641000
/sys/class/power_supply/BAT0/charge_full_design:9600000
/sys/class/power_supply/BAT0/charge_now:5431000
/sys/class/power_supply/BAT0/charge_type:Custom
/sys/class/power_supply/BAT0/current_now:0
/sys/class/power_supply/BAT0/cycle_count:0
grep: /sys/class/power_supply/BAT0/device: Is a directory
grep: /sys/class/power_supply/BAT0/hwmon1: Is a directory
/sys/class/power_supply/BAT0/manufacturer:Notebook
/sys/class/power_supply/BAT0/model_name:BAT
grep: /sys/class/power_supply/BAT0/power: Is a directory
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/serial_number:0001
/sys/class/power_supply/BAT0/status:Not charging
grep: /sys/class/power_supply/BAT0/subsystem: Is a directory
/sys/class/power_supply/BAT0/technology:Li-ion
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/uevent:DEVTYPE=power_supply
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_NAME=BAT0
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_TYPE=Battery
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_STATUS=Not charging
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_PRESENT=1
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_TECHNOLOGY=Li-ion
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CYCLE_COUNT=0
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_VOLTAGE_MIN_DESIGN=7700000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_VOLTAGE_NOW=7752000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CURRENT_NOW=0
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CHARGE_FULL_DESIGN=9600000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CHARGE_FULL=8641000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CHARGE_NOW=5431000
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CAPACITY=62
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_CAPACITY_LEVEL=Normal
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_MODEL_NAME=BAT
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_MANUFACTURER=Notebook
/sys/class/power_supply/BAT0/uevent:POWER_SUPPLY_SERIAL_NUMBER=0001
/sys/class/power_supply/BAT0/voltage_min_design:7700000
/sys/class/power_supply/BAT0/voltage_now:7752000
```
</details>

Minimum kernel version and user facing installation instructions are a more difficult question:

From what I have found, users are mostly expected to run TUXEDO OS or one of the hand full of distributions supported by Tomte, which automatically installs the drivers [^1][^2]. Apart from that, users can install the drivers from a package repository for their distribution or compile them from source[^3].

Unfortunately I have, up to now, failed to find a minimum required kernel version. I would expect something in the driver repository or on the TUXEDO website, however no luck. The control file in the `tuxedo-drivers.deb` from TUXEDO's .deb repository[^4] only contains the line `Depends: dkms (>= 2.1.0.0)`. Which I fear is not helpful, but I am not sure. Could you point me in a reasonable direction to find this information?

[^1]: https://www.tuxedocomputers.com/en/Downloads-Drivers.tuxedo
[^2]: https://www.tuxedocomputers.com/en/Infos/Help-Support/Frequently-asked-questions/What-is-TUXEDO-Tomte-.tuxedo
[^3]: https://github.com/AaronErhardt/tuxedo-rs?tab=readme-ov-file#tuxedo-driver-modules
[^4]: https://deb.tuxedocomputers.com/ubuntu/

linrunner, 2025-07-07:
@almndtr33 I have merged your PR, made some adjustments and added an automated test. The latter serves two purposes:

1. To test the logic and check the error messages on the real hardware --> your task.

2. To run it in simulation mode on a ThinkPad. This allows me to check the logic for regressions when all plugins are affected by a change or the calling code is change. This is very important for continous maintenance of battery care without having access to all the real hardware.

**Please run the automated test on your Tuxedo laptop:**
Install the tool **[clitest](https://github.com/aureliojargas/clitest)**. It should be available as a package in most distributions. 

Clone the **feature/bat.d/tuxedo** branch (note that I made forced pushes) and change to the **./unit-tests** directory. Then run the test script (with sudo or as root) and show the output:

     ./test-bc_tuxedo.sh

almndtr33, 2025-07-11:
The output after running the command as requested:

```
# ./test-bc_tuxedo.sh 
        # bata=BAT0 batb=BAT2
#1	# +++ Tuxedo laptops +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2	#
#3	# --- tlp start
#4	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}= STOP_CHARGE_THRESH_${bata}= START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}=
#5	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="60" STOP_CHARGE_THRESH_${bata}="100" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#6	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="100" STOP_CHARGE_THRESH_${bata}="100" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#7	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="50" STOP_CHARGE_THRESH_${bata}="0" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#8	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="50" STOP_CHARGE_THRESH_${bata}="101" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#9	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="80" STOP_CHARGE_THRESH_${bata}="80" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/g"
#10	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="80" STOP_CHARGE_THRESH_${bata}="90" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}=
#11	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="DEF" STOP_CHARGE_THRESH_${bata}="DEF" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}=
#12	sudo tlp start -- ${xinc} NATACPI_ENABLE=0 START_CHARGE_THRESH_${bata}="DEF" STOP_CHARGE_THRESH_${bata}="DEF"
#13	#
#14	# --- tlp setcharge w/o arguments
#15	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="60" STOP_CHARGE_THRESH_${bata}="100" X_SOC_CHECK=0 2>&1 | sed -r "s/${bata}/BATA/"
#16	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="100" STOP_CHARGE_THRESH_${bata}="100" 2>&1 | sed -r "s/${bata}/BATA/"
#17	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="50" STOP_CHARGE_THRESH_${bata}="0" 2>&1 | sed -r "s/${bata}/BATA/"
#18	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="40" STOP_CHARGE_THRESH_${bata}="85" 2>&1 | sed -r "s/${bata}/BATA/"
#19	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="45" STOP_CHARGE_THRESH_${bata}="90" 2>&1 | sed -r "s/${bata}/BATA/"
#20	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="80" STOP_CHARGE_THRESH_${bata}="70" 2>&1 | sed -r "s/${bata}/BATA/"
#21	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="70" STOP_CHARGE_THRESH_${bata}="90" 2>&1 | sed -r "s/${bata}/BATA/"
#22	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="70" STOP_CHARGE_THRESH_${bata}="90" 2>&1 | sed -r "s/${bata}/BATA/"
#23	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="DEF" STOP_CHARGE_THRESH_${bata}="DEF" 2>&1 | sed -r "s/${bata}/BATA/"
#24	sudo tlp setcharge -- ${xinc} NATACPI_ENABLE=0 START_CHARGE_THRESH_${bata}="DEF" STOP_CHARGE_THRESH_${bata}="DEF" 2>&1 | sed -r "s/${bata}/BATA/"
#25	#
#26	# --- tlp setcharge w/ arguments
#27	sudo tlp setcharge 60 100 -- ${xinc} X_SOC_CHECK=0 2>&1 | sed -r "s/${bata}/BATA/"
#28	sudo tlp setcharge 50 90 -- ${xinc} X_THRESH_SIMULATE_WRITEERR=1 X_SOC_CHECK=0 2>&1 | sed -r "s/${bata}/BATA/"
#29	sudo tlp setcharge 100 100 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#30	sudo tlp setcharge 50 0 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#31	sudo tlp setcharge 40 85 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#32	sudo tlp setcharge XYZZY 0 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#33	sudo tlp setcharge 50 XYZZY -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#34	sudo tlp setcharge 80 70 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#35	sudo tlp setcharge 70 90 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#36	sudo tlp setcharge 70 90 -- ${xinc} X_THRESH_SIMULATE_READERR="1" 2>&1 | sed -r "s/${bata}/BATA/"
#37	sudo tlp setcharge 70 90 -- ${xinc} X_THRESH_SIMULATE_START="60" X_THRESH_SIMULATE_STOP="100" 2>&1 | sed -r "s/${bata}/BATA/"
#38	sudo tlp setcharge 70 90 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#39	sudo tlp setcharge DEF DEF -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#40	sudo tlp setcharge ${batb} -- ${xinc} 2>&1 | sed -r "s/${batb}/BATB/"
#41	sudo tlp setcharge 0 3 ${batb} -- ${xinc} 2>&1 | sed -r "s/${batb}/BATB/"
#42	sudo tlp setcharge XYZZY ABCDE ${batb} -- ${xinc} 2>&1 | sed -r "s/${batb}/BATB/"
#43	exit

+++ Test Summary ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
charge-thresholds_tuxedo                           --> 
```

(I hope I followed your instructions correctly, it looks like something is missing after the arrow in the last line.)

linrunner, 2025-07-12:
@almndtr33 Looks pretty perfect already.

> it looks like something is missing after the arrow in the last line

An `exit` was left over from the development test phase. I have removed it. Please retest.

By the way: what do you think of using these Tuxedo-specific sysfiles instead of the hard-coded threshold value sets?
```
/sys/class/power_supply/BAT0/charge_control_end_available_thresholds:60 70 80 90 100
/sys/class/power_supply/BAT0/charge_control_start_available_thresholds:40 50 60 70 80 95
```

It's perfectionism, I know ...

linrunner, 2025-07-12:
When determining the available features, I tend to be pragmatic: the first battery found is enough for me and I generalize its charge control attributes to all batteries ([see line #113](https://github.com/linrunner/TLP/blob/feature/bat.d/tuxedo/bat.d/70-tuxedo#L113)). The vast majority of laptops only have one battery anyway. Basically, the loop via “BAT0 BAT1” in TEMPLATE is pure convenience, as this fits the vast majority of models.

I have implemented it: https://github.com/linrunner/TLP/commit/55bafc7272c6ad44fb26c71032c344cac6d043bc

Please run the automated test again and also show the output of

    tlp-stat -b

linrunner, 2025-07-13:
Great. Thanks for your contribution. Merged to main.

Do you perhaps have contacts with other Tuxedo users who could also test this? One can never have enough testers ... 
Packages for Arch are already available. I could provide them for Debian/Ubuntu.

TabError, 2025-07-24:
I also have a Tuxedo (Pulse 15 Gen2) and could help testing.
(But I'd need more precise instructions what exactly I am supposed to do.)

linrunner, 2025-07-25:
@TabError have you already installed **tuxedo-drivers**, as I told you in #819? If not, please do so now. Tuxedo provides detailed instructions here: https://www.tuxedocomputers.com/en/Add-TUXEDO-software-package-sources.tuxedo

After that show the output of:

    sudo modprobe -v --first-time clevo_acpi
    tlp-stat --psup

After you've installed the kernel module we have to use a little bit of trickery, as I can't provide alpha packages for Fedora ad hoc. First, we need TLP 1.8.0, as the Fedora maintainer didn't manage to get it ready for F42. Download it [here](https://rpmfind.net/linux/RPM/fedora/devel/rawhide/x86_64/t/tlp-1.8.0-1.fc43.noarch.html) and install it manually.

linrunner, 2025-07-27:
@almndtr33 _asus_wmi_ is a bit intrusive, therefore I moved the asus plugin to the end of the detection chain [with 1.7.0](https://github.com/linrunner/TLP/commit/5f4e3d8eb88490466f6c667d5362c94a83493058).

If TLP falls through to the asus plugin with 1.9.0 alpha, then _clevo_acpi_ may not be loaded. The tester should check this beforehand and also whether the sysfiles are there:

    sudo modprobe -v --first-time clevo_acpi
    tlp-stat --psup

If _clevo_acpi_ fails to load, it may not be a Clevo OEM chassis.

TabError, 2025-07-31:
I have installed tuxedo-drivers. (However this does not include [tuxedo tomte](https://www.tuxedocomputers.com/en/Infos/Help-Support/Frequently-asked-questions/What-is-TUXEDO-Tomte-.tuxedo), which is not supported for fedora. But this may not be so important here.)

The output looks like this:
```
> sudo modprobe -v --first-time clevo_acpi
modprobe: ERROR: could not insert 'clevo_acpi': Module already in kernel
> tlp-stat --psup
--- TLP 1.7.0 --------------------------------------------

+++ Power supply diagnostic
/sys/class/power_supply/AC0/type: Mains
/sys/class/power_supply/AC0/usb_type: (not available)
/sys/class/power_supply/AC0/online: 1
/sys/class/power_supply/AC0/voltage_max: (not available)
/sys/class/power_supply/AC0/voltage_min: (not available)
/sys/class/power_supply/AC0/voltage_now: (not available)
/sys/class/power_supply/AC0/present: (not available)
/sys/class/power_supply/AC0/charge_control_start_threshold: (not available)
/sys/class/power_supply/AC0/charge_control_end_threshold: (not available)
/sys/class/power_supply/AC0/status: (not available)
/sys/class/power_supply/AC0/device/path: (not available)
/sys/class/power_supply/BAT0/type: Battery
/sys/class/power_supply/BAT0/usb_type: (not available)
/sys/class/power_supply/BAT0/online: (not available)
/sys/class/power_supply/BAT0/voltage_max: (not available)
/sys/class/power_supply/BAT0/voltage_min: (not available)
/sys/class/power_supply/BAT0/voltage_now: 12572000
/sys/class/power_supply/BAT0/present: 1
/sys/class/power_supply/BAT0/charge_control_start_threshold: (not available)
/sys/class/power_supply/BAT0/charge_control_end_threshold: (not available)
/sys/class/power_supply/BAT0/status: Charging
/sys/class/power_supply/BAT0/device/path: \_SB_.BAT0

+++ udev diagnostic
Reading rules file: /usr/lib/udev/rules.d/85-tlp.rules
  RUN{program} : /usr/sbin/tlp auto
OK.
```

as per the download of tlp1.8.0 - I assume I am supposed to download it from [this link](https://download.fedoraproject.org/pub/fedora/linux/development/rawhide/Everything/source/tree/Packages/t/tlp-1.8.0-1.fc43.src.rpm)? Because it tells me 404 not found like the url is not present on the server. (If I had the rpm file, it would just be `dnf install path/to/file.rpm`, right?)

linrunner, 2025-08-01:
@TabError 

> sudo modprobe -v --first-time clevo_acpi
> modprobe: ERROR: could not insert 'clevo_acpi': Module already in kernel

Means that the module is loaded.

> /sys/class/power_supply/BAT0/charge_control_start_threshold: (not available)
> /sys/class/power_supply/BAT0/charge_control_end_threshold: (not available)

Means that nevertheless thresholds are not supported. 

> as per the download of tlp1.8.0 - I assume I am supposed to download it from [this [...]

TabError, 2025-08-02:
I just ran `tlp-stat --psup` and I got this
<details><summary>Output</summary>

```
> tlp-stat --psup 
--- TLP 1.8.0 --------------------------------------------

+++ Power supply diagnostic
/sys/class/power_supply/AC0/type: Mains
/sys/class/power_supply/AC0/online: 0
/sys/class/power_supply/BAT0/type: Battery
/sys/class/power_supply/BAT0/present: 1
/sys/class/power_supply/BAT0/capacity: 86
/sys/class/power_supply/BAT0/charge_full_design: 7900000
/sys/class/power_supply/BAT0/charge_full: 7900000
/sys/class/power_supply/BAT0/charge_now: 6794000
/sys/class/power_supply/BAT0/current_now: 1106000
/sys/class/power_supply/BAT0/voltage_now: 11808000
/sys/class/power_supply/BAT0/status: Discharging
/sys/class/power_supply/BAT0/device/path: \_SB_.BAT0

+++ udev diagnostic
Reading rules file: /usr/lib/udev/rules.d/85-tlp.rules
  RUN{program} : /usr/sbin/tlp auto
OK.
```
</details> 

This is good, right?

TabError, 2025-08-02:
> Did clevo_acpi still need to be loaded at this point, or was it already there?

It always says, `Module already in kernel` so i guess no.

> output of `ls -l /sys/module/clevo_acpi`

```
> ls -l /sys/module/clevo_acpi
total 0
drwxr-xr-x. 2 root root    0  2. Aug 17:38 drivers/
drwxr-xr-x. 2 root root    0  2. Aug 17:38 holders/
drwxr-xr-x. 2 root root    0  2. Aug 17:38 notes/
drwxr-xr-x. 2 root root    0  2. Aug 17:38 sections/
-r--r--r--. 1 root root 4096  2. Aug 17:38 coresize
-r--r--r--. 1 root root 4096  2. Aug 17:38 initsize
-r--r--r--. 1 root root 4096  2. Aug 17:38 initstate
-r--r--r--. 1 root root 4096  2. Aug 17:38 refcnt
-r--r--r--. 1 root root 4096  2. Aug 17:38 taint
--w-------. 1 root root 4096  2. Aug 17:38 uevent
```

> Did you download and install the plugin file afterwards as I wrote above?

Sry, haven't read that ...

<details><summary>Details</summary>

[image]

</details> 

I guess i should make it executable? (just done that)

> Then show the output of ...

<details><summary>Details</summary>

```
 jp  ~ 
 tail -3 /etc/tlp.conf
#DEVICES_TO_DISABLE_ON_UNDOCK=""

TLP_DEBUG="arg bat run"
 jp  ~ 
 sudo tlp noop
Debug: no operation performed.
Error: message color test.
Warning: message color test.
Notice: message color test.
Success: message color test.
 jp  ~ 
 sudo tlp-stat -T
--- TLP 1.8.0 --------------------------------------------

Aug 02 17:50:29 fedora.fritz.box tlp[44971]: parse_args4config: tlp noop --
Aug 02 17:50:29 fedora.fritz.box tlp[44971]: +++ noop (1.8.0) ++++++++++++++++++++++++++++++++++++++++
Aug 02 17:50:29 fedora.fritz.box tlp[44971]: SHELL=/usr/bin/bash; umask=0022
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: power_source=bat
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: manual_mode=n
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: power_mode=bat
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: check_thinkpad.not_a_thinkpad: model=Standard
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: batdrv_init.thinkpad.not_a_thinkpad
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: check_thinkpad.not_a_thinkpad: model=Standard
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: batdrv_init.thinkpad-legacy.not_a_thinkpad
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: batdrv_init.lenovo.no_match
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: batdrv_init.huawei.no_match
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: batdrv_init.msi.no_match
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: batdrv_init.samsung.no_match
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: batdrv_init.lg.no_match
Aug 02 17:50:30 fedora.fritz.box tlp[44971]: batdrv_init.lg-legacy.no_match
[... 93 lines cut ...]
State          = enabled
RDW state      = not installed
Last run       = unknown
Mode           = unknown
Power source   = battery

+++ Battery Care
Plugin: tuxedo
Supported features: none available
Driver usage:
* natacpi (clevo_acpi) = inactive (laptop not supported)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = OEM
/sys/class/power_supply/BAT0/model_name                     = standard
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   7900 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   7900 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   3634 [mAh]
/sys/class/power_supply/BAT0/current_now                    =   1038 [mA]
/sys/class/power_supply/BAT0/status                         = Discharging

Charge                                                      =   46.0 [%]
Capacity                                                    =  100.0 [%]```
</details>

TabError, 2025-08-02:
I'm sorry, I had it wrong. I just powered my laptop off and now on again. when i ran `tlp-stat -s -b` it showed
```
+++ Battery Care
Plugin: asus
Supported features: none available
Driver usage:
* natacpi (asus_wmi) = inactive (laptop not supported)
```
only after executing `sudo modprobe -v --first-time clevo_acpi` again it read
```
+++ Battery Care
Plugin: tuxedo
Supported features: none available
Driver usage:
* natacpi (clevo_acpi) = inactive (laptop not supported)
```

so i guess it is not loaded automatically...

linrunner, 2025-08-02:
@TabError To load automatically on boot create a file **/etc/modules-load.d/tlp.conf** containing the line

    clevo_acpi

linrunner, 2025-08-04:
@moriglia thanks for your report. 

@almndtr33 Regarding your Tuxedo, was _clevo_acpi_ loaded automatically, or did you have to intervene?

almndtr33, 2025-08-05:
I had to install the tuxedo-drivers, but that already happened some while ago to support keyboard backlight and so on. The laptop is running NixOS so I just used the [`hardware.tuxedo-drivers.enable`](https://search.nixos.org/options?channel=25.05&show=hardware.tuxedo-drivers.enable) option. As far as I can tell, under the hood this option ensures that the package from Tuxedos GitLab is installed and added to the kernel.

I just verified that `tlp` uses the generic plugin if `hardware.tuxedo.enable` is not set. In this case there are also no start and end threshold files in `/sys/clall/power_supply/BAT0`.

If `hardware.tuxedo.enable` **is** set, running `rmmod -v clevo_acpi` causes TLP to use the generic driver and removes the charge threshold sysfiles. Subsequently running `modprobe -v --first-time clevo_acpi` restores my previous working state with TLP using the tuxedo plugin and the charge threshold files in place.

moriglia, 2025-08-08:
>     1. Check whether `cat /sys/devices/virtual/dmi/id/sysvendor` = “TUXEDO”  - does the string match?
>        [@moriglia](https://github.com/moriglia) [@TabError](https://github.com/TabError) Does this also match on your hardware?

I don't have a `sysvendor` file, however I do have a `chassis_vendor` and a `sys_vendor` files, and both match "TUXEDO":
```bash
$ ll /sys/devices/virtual/dmi/id/*vendor
[...]
-r--r--r-- 1 root root 4096 Aug  8 10:32 /sys/devices/virtual/dmi/id/chassis_vendor
-r--r--r-- 1 root root 4096 Aug  8 10:24 /sys/devices/virtual/dmi/id/sys_vendor
$ cat /sys/devices/virtual/dmi/id/*vendor
[...]
TUXEDO
TUXEDO
```

linrunner, 2025-08-08:
@moriglia that's exactly what I need. Thank you.

almndtr33, 2025-08-10:
> I still don't know whether the kernel loads _clevo_acpi_ on its own based on hardware detection, or whether this is enforced by an additional configuration file.

From what I understand, this option forces the `tuxedo_keyboard` kernel module to be loaded at boot. The [manual](https://nixos.org/manual/nixos/stable/#sec-kernel-config) explicitly contrasts this to the statement that "modules for hardware devices are generally loaded automatically by `udev`". So it is not loaded based on hardware detection on my machine.

I did some additional tests on an old TUXEDO OS partition on my laptop. When I initially booted it, `clevo_acpi` was not loaded and I could also not edit charge thresholds in the Tuxedo Control Center (TCC). Initially, loading the module with `modprobe` failed (module not found) but after installing some updates, it magically worked (and also inserted two other modules on its own). After this, charge thresholds where available in the sysfs files, in the TCC and with TLP. I could not reproduce the state where `clevo_acpi` was not available. Even booting from a TUXEDO OS installer USB stick, `clevo_acpi` was immediately loaded and charge thresholds were available.

The only interpretation I can think of, is that my hardware is supported while the other two are not. However, this seems improbable as @moriglia has an InfinityBook Pro Gen7 which I would expect to be very similar to my InfinityBook S Gen8.

linrunner, 2025-08-11:
> InfinityBook Pro Gen7 which I would expect to be very similar to my InfinityBook S Gen8

That's just an assumption at this stage.

Out-of-tree drivers can cause a lot of trouble and work, as I learned over 15 years with ThinkPads. But at least the hardware was clearly defined there. 

With Tuxedo  there is “something undefined” in every box - based on the limited information available so far. On the other hand, there aren't that many of them. @TabError @moriglia have you asked support about your model yet?

I think I will implement the differentiated approach and we'll see.

TabError, 2025-08-15:
<details><summary>Details</summary>

```
> wget https://raw.githubusercontent.com/linrunner/TLP/refs/heads/bat.d/tuxedo/bat.d/70-tuxedo
Saving '70-tuxedo'
HTTP response 200  [https://raw.githubusercontent.com/linrunner/TLP/refs/heads/bat.d/tu70-tuxedo
                     100% [====================================>]    6.11K    --.-KB/s
                          [Files: 1  Bytes: 6.11K [26.34KB/s] Re]
> l 70*
.rw-r--r--@ 27k jp 15 Aug 04:11 70-tuxedo
> mv 70-tuxedo /usr/share/tlp/bat.d/
mv: replace '/usr/share/tlp/bat.d/70-tuxedo', overriding mode 0755 (rwxr-xr-x)? y
mv: inter-device move failed: '70-tuxedo' to '/usr/share/tlp/bat.d/70-tuxedo'; unable to remove target: Permission denied
> chmod a+x 70-tuxedo 
> sudo mv 70-tuxedo /usr/share/tlp/bat.d/
[sudo] password for jp: 
> l /usr/share/tlp/bat.d/
.rwxr-xr-x@  50k root 25 Jul 02:00 05-thinkpad
.rwxr-xr-x@  42k root 25 Jul 02:00 10-thinkpad-legacy
.rwxr-xr-x@  17k root 25 Jul 02:00 15-lenovo
.rwxr-xr-x@  20k root 25 Jul 02:00 20-huawei
.rwxr-xr-x@  22k root 25 Jul 02:00 25-msi
.rwxr-xr-x@  19k root 25 Jul 02:00 30-samsung
.rwxr-xr-x@  20k root 25 Jul 02:00 35-lg
.rwxr-xr-x@  19k root 25 Jul 02:00 36-lg-legacy
.rwxr-xr-x@  19k root 25 Jul 02:00 40-sony
.rwxr-xr-x@  24k root 25 Jul 02:00 45-system76
.rwxr-xr-x@  20k root 25 Jul 02:00 50-toshiba
.rwxr-xr-x@  45k root 25 Jul 02:00 55-cros-ec
.rwxr-xr-x@  20k root 25 Jul 02:00 56-framework
.rwxr-xr-x@  24k root 25 Jul 02:00 60-macbook
.rwxr-xr-x@  26k root 25 Jul 02:00 65-dell
.rwxr-xr-x@  27k jp   15 Aug 04:11 70-tuxedo
.rwxr-xr-x@  20k root 25 Jul 02:00 89-asus
.rwxr-xr-x@ 8.0k root 25 Jul 02:00 90-generic
.rwxr-xr-x@  25k root 25 Jul 02:00 TEMPLATE
> sudo modprobe -rv clevo_apci
modprobe: FATAL: Module clevo_apci not found.
> sudo modprobe -rv clevo_acpi
rmmod clevo_acpi
> sudo modprobe -rv clevo_acpi
> sudo tlp-stat -b
--- TLP 1.8.0 --------------------------------------------

+++ Battery Care
Plugin: tuxedo
Supported features: none available
Driver usage:
* natacpi (clevo_acpi) = inactive (Tuxedo model not supported)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = OEM
/sys/class/power_supply/BAT0/model_name                     = standard
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   7900 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   7900 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   4029 [mAh]
/sys/class/power_supply/BAT0/current_now                    =   1922 [mA]
/sys/class/power_supply/BAT0/status                         = Discharging

Charge                                                      =   51.0 [%]
Capacity                                                    =  100.0 [%]
```
</details>

linrunner, 2025-08-15:
I just added another improvement: `tlp-stat -b` output on unsupported models will show

> /sys/class/power_supply/BAT0/charge_control_start_threshold =  **(not available)**
> /sys/class/power_supply/BAT0/charge_control_end_threshold   =  **(not available)**
> /sys/class/power_supply/BAT0/charge_type                    = **(not available)**

to clarify where the problem stems from.

almndtr33, 2025-08-15:
Built from commit `e1ed9ae18ecbb3bfca575846cb45f6ab0997d579` and everything appears to work well:
<details>
  <summary><tt>tlp-stat -s -b</tt></summary>
  
 
  ```
$ sudo tlp-stat -b -s
--- TLP 1.9.0-alpha.0 --------------------------------------------

+++ System Info
System         = TUXEDO Not Applicable TUXEDO InfinityBook S Gen8
BIOS           = 1.07.05RTR3 INSYDE Corp.
EC Firmware    = 7.2
OS Release     = NixOS 25.05 (Warbler)
Kernel         = 6.12.38 #1-NixOS SMP PREEMPT_DYNAMIC Mon Jul 14 14:02:59 UTC 2025 x86_64
/proc/cmdline  = initrd=\EFI\nixos\6kmdkfs6bah14450lbcfkp788qmw36yg-initrd-linux-6.12.38-initrd.efi init=/nix/store/5c34wf9a9d86pkdbvryz6bnf776whd1l-nixos-system-knight-nixos-25.05.20250715.32a4e87/init loglevel=4 lsm=landlock,yama,bpf
Init system    = systemd 257
Boot mode      = UEFI
Suspend mode   = [s2idle] deep

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 09:04:04 PM, 254 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: tuxedo
Supported features: charge thresholds
Driver usage:
* natacpi (clevo_acpi) = active (charge thresholds)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  40/50/60/70/80/95(default)
* STOP_CHARGE_THRESH_BAT0/1:   60/70/80/90/100(default)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = Notebook
/sys/class/power_supply/BAT0/model_name                     = BAT
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   9600 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   8641 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   5952 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Not charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     60 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     70 [%]
/sys/class/power_supply/BAT0/charge_type                    = Custom

Charge                                                      =   68.9 [%]
Capacity                                                    =   90.0 [%]
  ```
  
</details>

However, one of the tests in `unit-tests/test-bc_tuxedo.sh` fails:
<details>
  <summary><tt>test-bc_tuxedo.sh</tt></summary>
 
  ```
./test-bc_tuxedo.sh 
        # bata=BAT0 batb=BAT2
#1	# +++ Tuxedo laptops +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2	#
#3	# --- tlp start
#4	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}= STOP_CHARGE_THRESH_${bata}= START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}=
#5	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="60" STOP_CHARGE_THRESH_${bata}="100" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#6	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="100" STOP_CHARGE_THRESH_${bata}="100" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#7	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="50" STOP_CHARGE_THRESH_${bata}="0" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
[... 42 lines cut ...]
-/sys/class/power_supply/BAT0/charge_control_start_threshold = (not available) [%]
-/sys/class/power_supply/BAT0/charge_control_end_threshold   = (not available) [%]
+/sys/class/power_supply/BAT0/charge_control_start_threshold =     95 [%]
+/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
#47	#
#48	# --- Reset test machine to configured thresholds
#49	sudo tlp setcharge ${bata} - ${xinc} > /dev/null 2>&1
#50	#

FAIL: 1 of 50 tests failed

+++ Test Summary ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
charge-thresholds_tuxedo                           --> FAIL: 1 of 50 tests failed
  ```
  
</details>

I also did some more research. Most notably I found [this](https://github.com/AaronErhardt/tuxedo-rs/issues/80#issuecomment-2058968855) comment in a `tuxedo-rs` issue (which I already mentioned in my first message). Apparently there are two charge control configurations: One where a profile ('high capacity' -> 100%, 'balanced' -> 90% or 'stationary' -> 80%)[^1] can be chosen, which automatically sets predefined charge thresholds in `/sys/devices/platform/tuxedo_keyboard/charging_profile/charging_profile`. From what I can tell, this is the configuration for uniwill chargers which use the driver [uniwill_wmi](https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers/-/blob/main/src/uniwill_keyboard.h#L649). The other configuration is the one on my laptop, i.e. `clevo_acpi` and explicit charge thresholds. (According to the `tuxedo-rs` issue there is a third hardware vendor 'schenker', however I could find nothing relating to schenker in the `tuxedo-drivers` code.)

My system only loads the module `tuxedo_keyboard` explicitly, so this appears to be the 'root' module that loads others. If I run `lsmod | grep tuxedo_keyboard` I see the line `tuxedo_keyboard       114688  3 clevo_acpi,tuxedo_io,clevo_wmi` (among others). When I asked @moriglia to run the same command, no `clevo_acpi` was in the output but `u [... cut ...] and the `tuxedo-drivers` code well enough to figure out how it decides to load the clevo or uniwill module.

TLP apparently has support for some laptops (toshiba, sony, samsung, lenovo, ...) with charge control that works in preset configurations like for the uniwill charger. If there is interest, I could try writing a plugin for the uniwill charger. However, I could not test it on my own.

[^1]: https://www.tuxedocomputers.com/en/Battery-charging-profiles-inside-the-TUXEDO-Control-Center.tuxedo

linrunner, 2025-08-16:
@almndtr33 test fixed.
@moriglia packages updated.

@TabError @moriglia `tlp-stat -b` message changed to  _"Tuxedo's non-Clevo models are unsupported"_. All based on current knowledge.

almndtr33, 2025-08-16:
Everything looks fine now:

<details>
  <summary>Console output</summary>
  
  ```
$ sudo tlp-stat -b -s
--- TLP 1.9.0-alpha.0 --------------------------------------------

+++ System Info
System         = TUXEDO Not Applicable TUXEDO InfinityBook S Gen8
BIOS           = 1.07.05RTR3 INSYDE Corp.
EC Firmware    = 7.2
OS Release     = NixOS 25.05 (Warbler)
Kernel         = 6.12.38 #1-NixOS SMP PREEMPT_DYNAMIC Mon Jul 14 14:02:59 UTC 2025 x86_64
/proc/cmdline  = initrd=\EFI\nixos\6kmdkfs6bah14450lbcfkp788qmw36yg-initrd-linux-6.12.38-initrd.efi init=/nix/store/5c34wf9a9d86pkdbvryz6bnf776whd1l-nixos-system-knight-nixos-25.05.20250715.32a4e87/init loglevel=4 lsm=landlock,yama,bpf
Init system    = systemd 257
Boot mode      = UEFI
Suspend mode   = [s2idle] deep

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 03:26:53 PM, 8 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: tuxedo
Supported features: charge thresholds
Driver usage:
* natacpi (clevo_acpi) = active (charge thresholds)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  40/50/60/70/80/95(default)
* STOP_CHARGE_THRESH_BAT0/1:   60/70/80/90/100(default)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = Notebook
/sys/class/power_supply/BAT0/model_name                     = BAT
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   9600 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   8641 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   5956 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Not charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     60 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =     70 [%]
/sys/class/power_supply/BAT0/charge_type                    = Custom

Charge                                                      =   68.9 [%]
Capacity                                                    =   90.0 [%]

$ ./test-bc_tuxedo.sh 
        # bata=BAT0 batb=BAT2
#1	# +++ Tuxedo laptops +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
#2	#
#3	# --- tlp start
#4	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}= STOP_CHARGE_THRESH_${bata}= START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}=
#5	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="60" STOP_CHARGE_THRESH_${bata}="100" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#6	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="100" STOP_CHARGE_THRESH_${bata}="100" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#7	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="50" STOP_CHARGE_THRESH_${bata}="0" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#8	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="50" STOP_CHARGE_THRESH_${bata}="101" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/"
#9	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="80" STOP_CHARGE_THRESH_${bata}="80" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}= 2>&1 | sed -r "s/${bata}/BATA/g"
#10	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="80" STOP_CHARGE_THRESH_${bata}="90" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}=
#11	sudo tlp start -- ${xinc} START_CHARGE_THRESH_${bata}="DEF" STOP_CHARGE_THRESH_${bata}="DEF" START_CHARGE_THRESH_${batb}= STOP_CHARGE_THRESH_${batb}=
#12	sudo tlp start -- ${xinc} NATACPI_ENABLE=0 START_CHARGE_THRESH_${bata}="DEF" STOP_CHARGE_THRESH_${bata}="DEF"
#13	#
#14	# --- tlp setcharge w/o arguments
#15	sudo tlp setcharge -- ${xinc} START_CHARGE_THRESH_${bata}="60" STOP_CHARGE_THRESH_${bata}="100" X_SOC_CHECK=0 2>&1 | sed -r "s/${bata}/BATA/"
[... 17 lines cut ...]
#33	sudo tlp setcharge 50 XYZZY -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#34	sudo tlp setcharge 80 70 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#35	sudo tlp setcharge 70 90 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#36	sudo tlp setcharge 70 90 -- ${xinc} X_THRESH_SIMULATE_READERR="1" 2>&1 | sed -r "s/${bata}/BATA/"
#37	sudo tlp setcharge 70 90 -- ${xinc} X_THRESH_SIMULATE_START="60" X_THRESH_SIMULATE_STOP="100" 2>&1 | sed -r "s/${bata}/BATA/"
#38	sudo tlp setcharge 70 90 -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#39	sudo tlp setcharge DEF DEF -- ${xinc} 2>&1 | sed -r "s/${bata}/BATA/"
#40	sudo tlp setcharge ${batb} -- ${xinc} 2>&1 | sed -r "s/${batb}/BATB/"
#41	sudo tlp setcharge 0 3 ${batb} -- ${xinc} 2>&1 | sed -r "s/${batb}/BATB/"
#42	sudo tlp setcharge XYZZY ABCDE ${batb} -- ${xinc} 2>&1 | sed -r "s/${batb}/BATB/"
#43	#
#44	# --- tlp-stat
#45	sudo tlp-stat -b -- ${xinc} | grep -E 'BAT0/charge_(control|behaviour)'
#46	sudo tlp-stat -b -- ${xinc} X_THRESH_SIMULATE_READERR=1 | grep -E 'BAT0/charge_(control|behaviour)'
#47	#
#48	# --- Reset test machine to configured thresholds
#49	sudo tlp setcharge ${bata} - ${xinc} > /dev/null 2>&1
#50	#
OK: 50 of 50 tests passed

+++ Test Summary ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
charge-thresholds_tuxedo                           --> OK: 50 of 50 tests passed
  ```
  
</details>

TabError, 2025-08-24:
<details><summary>Here the output on my system</summary>

```
> sudo modprobe -rv clevo_acpi
rmmod clevo_acpi
> sudo tlp-stat -s -b
--- TLP 1.8.0 --------------------------------------------

+++ System Info
System         = TUXEDO Standard TUXEDO Pulse 15 Gen2
BIOS           = N.1.06A06
EC Firmware    = 1.14
OS Release     = Fedora Linux 42 (KDE Plasma Desktop Edition)
Kernel         = 6.15.10-200.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Aug 15 15:57:06 UTC 2025 x86_64
/proc/cmdline  = BOOT_IMAGE=(hd0,gpt6)/vmlinuz-6.15.10-200.fc42.x86_64 root=UUID=eada8cae-c478-4332-bd2f-b5e73a9a72c8 ro rootflags=subvol=root resume=UUID=921cffce-acac-4511-843f-40fd6efd1451 rd.luks.uuid=luks-890bf457-cfa6-4c49-9d6f-816d92c074c6 rhgb quiet
Init system    = systemd 
Boot mode      = UEFI
Suspend mode   = [s2idle]
SELinux status = enabled (enforcing)

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 18:08:56, 6886 sec(s) ago
Mode           = battery
Power source   = battery

+++ Battery Care
Plugin: tuxedo
Supported features: none available
Driver usage:
* natacpi (clevo_acpi) = inactive (Tuxedo's non-Clevo models are unsupported)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = OEM
/sys/class/power_supply/BAT0/model_name                     = standard
/sys/class/power_supply/BAT0/cycle_count                    =      0 (or not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   7900 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   7900 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   3397 [mAh]
/sys/class/power_supply/BAT0/current_now                    =    783 [mA]
/sys/class/power_supply/BAT0/status                         = Discharging

                                                            = (not available) [%]
                                                            = (not available) [%]

Charge                                                      =   43.0 [%]
Capacity                                                    =  100.0 [%]```
</details>

linrunner, 2025-08-25:
@TabError thanks. I'lll take that into account for the documentation.