linrunner/TLP #804: Add support for systemd-netword

akrram09, 2025-06-15:
**Is your feature request related to a problem? Please describe.**
Yes, I use systemd-netword with iwd for wifi and when i force remove NetworkManager I get a lot of error messages when I check error for arch

**Describe the solution you'd like**
Add support for systemd-netword and make NetworkManager as optional dependencies that u can remove

akrram09, 2025-06-15:
> **Is your feature request related to a problem? Please describe.** Yes, I use systemd-netword with iwd for wifi, and when i force remove NetworkManager I get a lot of error messages when I check [...]

linrunner, 2025-06-16:
I don't know if you are aware that TLP consists of two packages:

1. **tlp** - does _not_ depend on NetworkManager
2. **tlp-rdw** - required for [RDW features](https://linrunner.de/tlp/settings/rdw.html), depends on NetworkManager

So if you don't want NetworkManager, do install **tlp** only.

However, systemd-networkd does not provide all the functionality required for RDW. So if you need RDW features, there is no way to  around the **tlp-rdw** package an Networkmanager.

Unfortunately, you do not show the outputs required by the issue template, nor do you mention your Distribution.

linrunner, 2025-06-17:
Let's see what features you are actually using. Please show the output of

    tlp-stat -c

akrram09, 2025-06-17:
`akram@arch ~ $ tlp-stat -c
--- TLP 1.8.0 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0038: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
defaults.conf L0008: TLP_MSG_COLORS="91 93 1 92"
/etc/tlp.conf L0072: TLP_PERSISTENT_DEFAULT="1"
defaults.conf L0010: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0011: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0012: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0013: MAX_LOST_WORK_SECS_ON_BAT="60"
/etc/tlp.conf L0152: CPU_ENERGY_PERF_POLICY_ON_AC="performance"
/etc/tlp.conf L0153: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
/etc/tlp.conf L0191: NMI_WATCHDOG="0"
/etc/tlp.conf L0222: DISK_DEVICES="nvme0n1"
/etc/tlp.conf L0230: DISK_APM_LEVEL_ON_AC="254 254"
/etc/tlp.conf L0231: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0020: DISK_APM_CLASS_DENYLIST="usb ieee1394"
/etc/tlp.conf L0270: DISK_IOSCHED="none"
/etc/tlp.conf L0278: SATA_LINKPWR_ON_AC=""
/etc/tlp.conf L0279: SATA_LINKPWR_ON_BAT=""
/etc/tlp.conf L0297: AHCI_RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0298: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0026: AHCI_RUNTIME_PM_TIMEOUT="15"
/etc/tlp.conf L0389: PCIE_ASPM_ON_AC="default"
/etc/tlp.conf L0390: PCIE_ASPM_ON_BAT="default"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0030: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_AC="default" #! Parameter is deprecated: feature will be removed with the next release
defaults.conf L0032: RADEON_POWER_PROFILE_ON_BAT="default" #! Parameter is deprecated: feature will be removed with the next release
defaults.conf L0033: AMDGPU_ABM_LEVEL_ON_AC="0"
defaults.conf L0034: AMDGPU_ABM_LEVEL_ON_BAT="1"
/etc/tlp.conf L0361: WIFI_PWR_ON_AC="off"
/etc/tlp.conf L0362: WIFI_PWR_ON_BAT="on"
defaults.conf L0037: WOL_DISABLE="Y"
/etc/tlp.conf L0375: SOUND_POWER_SAVE_ON_AC="10"
/etc/tlp.conf L0376: SOUND_POWER_SAVE_ON_BAT="10"
defaults.conf L0040: SOUND_POWER_SAVE_CONTROLLER="Y"
/etc/tlp.conf L0312: BAY_POWEROFF_ON_AC="0"
/etc/tlp.conf L0313: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0043: BAY_DEVICE="sr0"
/etc/tlp.conf L0395: RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0396: RUNTIME_PM_ON_BAT="auto"
/etc/tlp.conf L0413: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon xhci_hcd"
/etc/tlp.conf L0427: USB_AUTOSUSPEND="1"
/etc/tlp.conf L0441: USB_EXCLUDE_AUDIO="1"
/etc/tlp.conf L0447: USB_EXCLUDE_BTUSB="1"
/etc/tlp.conf L0453: USB_EXCLUDE_PHONE="0"
/etc/tlp.conf L0459: USB_EXCLUDE_PRINTER="1"
/etc/tlp.conf L0465: USB_EXCLUDE_WWAN="0"
defaults.conf L0053: RESTORE_DEVICE_STATE_ON_STARTUP="0"
/etc/tlp.conf L0541: RESTORE_THRESHOLDS_ON_BAT="1"
/etc/tlp.conf L0074: NATACPI_ENABLE="0"
/etc/tlp.conf L0075: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0066: TLP_DEFAULT_MODE="BAT"
/etc/tlp.conf L0103: CPU_DRIVER_OPMODE_ON_AC="active"
/etc/tlp.conf L0104: CPU_DRIVER_OPMODE_ON_BAT="active"
/etc/tlp.conf L0122: CPU_SCALING_GOVERNOR_ON_AC="performance"
/etc/tlp.conf L0123: CPU_SCALING_GOVERNOR_ON_BAT="powersave"
/etc/tlp.conf L0161: CPU_MIN_PERF_ON_AC="0"
/etc/tlp.conf L0162: CPU_MAX_PERF_ON_AC="100"
/etc/tlp.conf L0163: CPU_MIN_PERF_ON_BAT="0"
/etc/tlp.conf L0164: CPU_MAX_PERF_ON_BAT="30"
/etc/tlp.conf L0173: CPU_BOOST_ON_AC="1"
/etc/tlp.conf L0174: CPU_BOOST_ON_BAT="1"
/etc/tlp.conf L0184: CPU_HWP_DYN_BOOST_ON_AC="1"
/etc/tlp.conf L0185: CPU_HWP_DYN_BOOST_ON_BAT="1"
/etc/tlp.conf L0201: PLATFORM_PROFILE_ON_AC="performance"
/etc/tlp.conf L0202: PLATFORM_PROFILE_ON_BAT="balanced"
/etc/tlp.conf L0214: MEM_SLEEP_ON_AC="s2idle"
/etc/tlp.conf L0215: MEM_SLEEP_ON_BAT="deep"
/etc/tlp.conf L0252: DISK_SPINDOWN_TIMEOUT_ON_AC="0 0"
/etc/tlp.conf L0253: DISK_SPINDOWN_TIMEOUT_ON_BAT="0 0"
/etc/tlp.conf L0325: INTEL_GPU_MIN_FREQ_ON_AC="300"
/etc/tlp.conf L0326: INTEL_GPU_MIN_FREQ_ON_BAT="300"
/etc/tlp.conf L0327: INTEL_GPU_MAX_FREQ_ON_AC="1100"
/etc/tlp.conf L0328: INTEL_GPU_MAX_FREQ_ON_BAT="800"
/etc/tlp.conf L0329: INTEL_GPU_BOOST_FREQ_ON_AC="1100"
/etc/tlp.conf L0330: INTEL_GPU_BOOST_FREQ_ON_BAT="800"
/etc/tlp.conf L0485: DEVICES_TO_DISABLE_ON_STARTUP="bluetooth wifi wwan"
/etc/tlp.conf L0491: DEVICES_TO_ENABLE_ON_STARTUP="wifi"
/etc/tlp.conf L0507: DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE="bluetooth wifi wwan"
/etc/tlp.conf L0532: START_CHARGE_THRESH_BAT0="75"
/etc/tlp.conf L0533: STOP_CHARGE_THRESH_BAT0="80"
/etc/tlp.conf L0535: START_CHARGE_THRESH_BAT1="75"
/etc/tlp.conf L0536: STOP_CHARGE_THRESH_BAT1="80"
/etc/tlp.conf L0552: DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi wwan"
/etc/tlp.conf L0553: DEVICES_TO_DISABLE_ON_WIFI_CONNECT="wwan"
/etc/tlp.conf L0554: DEVICES_TO_DISABLE_ON_WWAN_CONNECT="wifi"
`

linrunner, 2025-06-19:
These work without tlp-rdw/NetworkManager:
> /etc/tlp.conf L0485: DEVICES_TO_DISABLE_ON_STARTUP="bluetooth wifi wwan"
> /etc/tlp.conf L0491: DEVICES_TO_ENABLE_ON_STARTUP="wifi"
> /etc/tlp.conf L0507: DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE="bluetooth wifi wwan"

But these need tlp-rdw/NetworkManager to detect interface up/down:
> /etc/tlp.conf L0552: DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi wwan"
> /etc/tlp.conf L0553: DEVICES_TO_DISABLE_ON_WIFI_CONNECT="wwan"
> /etc/tlp.conf L0554: DEVICES_TO_DISABLE_ON_WWAN_CONNECT="wifi"

Even if systemd-networkd had the necessary capabilities, I wouldn't want to add a parallel implementation for it. Sorry about that.

OJFord, 2026-01-06:
@akrram09 In case you're still interested in this, I've created https://github.com/OJFord/tlp-rdw-systemd (AUR: https://aur.archlinux.org/packages/tlp-rdw-systemd) to sort of simulate the tlp-rdw behaviour on link (dis)connection on systemd-networkd systems without NetworkManager.

Do let me know (over there) if you try it and have any issues, or the setup instructions aren't clear. 🙂