linrunner/TLP #807: [ThinkPad X13s Gen 1] tlp-stat -b does not report battery information correctly

felixbohnacker, 2025-06-17:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

When running sudo tlp-stat -b, the following output is returned:
```
--- TLP 1.8.0 --------------------------------------------

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: qcom-battmgr-bat
/sys/class/power_supply/QCOM-BATTMGR-BAT/manufacturer       = (not available)
/sys/class/power_supply/QCOM-BATTMGR-BAT/model_name         = (not available)
/sys/class/power_supply/QCOM-BATTMGR-BAT/cycle_count        = (not supported)
/sys/class/power_supply/QCOM-BATTMGR-BAT/status             = (not available)

/sys/class/power_supply/QCOM-BATTMGR-BAT/charge_control_start_threshold = (not available) 
/sys/class/power_supply/QCOM-BATTMGR-BAT/charge_control_end_threshold = (not available) 
/sys/class/power_supply/QCOM-BATTMGR-BAT/charge_behaviour   = (not available)
```
The problem seems to be that this is case sensitive as I can only find **/sys/class/power_supply/qcom-battmgr-bat** on my system. 

`cat /sys/class/power_supply/qcom-battmgr-bat/cycle_count` returns the correct cycle count (3) though. The same is true for most of the other information (manufacturer, model name, status etc.).

Using the following bash script, I exported all the file contents within that directory:
```
for file in /sys/class/power_supply/qcom-battmgr-bat/*; do
    echo "$file:"
    cat "$file"
    echo
done
```
This can be found here: https://gist.github.com/felixbohnacker/4a7210a0c457d7102a4f33106445e414#file-gistfile1-txt

**Expected behavior**

All available battery information is reported when running tlp-stat -b.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
Both
2. Actions to reproduce the behaviour
Run sudo tlp-stat -b
3. Shell commands entered and their output *including error message(s)*
See above
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1** (not as file attachment, no screenshots)
https://gist.github.com/felixbohnacker/4ade5b79731d6b973886df5df599cf92#file-gistfile1-txt

**Additional context**

Add any other context about the problem here.

felixbohnacker, 2025-06-17:
Thank you very much, I commented out the line that converts the battery name to uppercase and started TLP again. This seems to have fixed the issue:
```
--- TLP 1.8.0 --------------------------------------------

+++ Battery Care
Plugin: generic
Supported features: none available

+++ Battery Status: qcom-battmgr-bat
/sys/class/power_supply/qcom-battmgr-bat/manufacturer       =SMP
/sys/class/power_supply/qcom-battmgr-bat/model_name         = LNV-5B10W51878
/sys/class/power_supply/qcom-battmgr-bat/cycle_count        =      3
/sys/class/power_supply/qcom-battmgr-bat/energy_full_design =  49530 [mWh]
/sys/class/power_supply/qcom-battmgr-bat/energy_full        =  50230 [mWh]
/sys/class/power_supply/qcom-battmgr-bat/energy_now         =  50230 [mWh]
/sys/class/power_supply/qcom-battmgr-bat/power_now          =      0 [mW]
/sys/class/power_supply/qcom-battmgr-bat/status             = Not charging

/sys/class/power_supply/qcom-battmgr-bat/charge_control_start_threshold = (not available) 
/sys/class/power_supply/qcom-battmgr-bat/charge_control_end_threshold = (not available) 
/sys/class/power_supply/qcom-battmgr-bat/charge_behaviour   = (not available) 

Charge                                                      =  100.0 [%]
Capacity                                                    =  101.4 [%]
```

By the way: ARM, not AMD :)

linrunner, 2025-06-17:
Thanks for reporting and testing.

> By the way: ARM, not AMD :)

Indeed ;-)

felixbohnacker, 2025-06-17:
Thank you very much for the quick reply and the great work overall!

closed 2025-06-17 by linrunner