linrunner/TLP #808: NFC is not disabled when configured to do so

Coacher, 2025-06-17:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

tlp is configured to disable NFC on boot and on battery, but NFC is not disabled.

**Expected behavior**

NFC is listed as blocked in `rfkill list` output after tlp start/tlp bat.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Configure tlp to disable NFC:
```
# tlp-stat --config | grep DEVICES_TO
/etc/tlp.conf L0487: DEVICES_TO_DISABLE_ON_STARTUP="nfc"
/etc/tlp.conf L0509: DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE="bluetooth nfc"
``` 
2. Boot the machine or run `tlp start`/`tlp bat`
3. Observe NFC is not blocked via rfkill:
```
# tlp start
TLP started in AC mode (auto).
# tlp-stat --rfkill
--- TLP 1.7.0 --------------------------------------------

+++ Wireless
bluetooth = on
nfc       = on
wifi      = on
wwan      = none (no device)

hci0(btusb)                   : bluetooth, connected
wlp0s20f3(iwlwifi)            : wifi, connected, power management = on

# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: nfc0: NFC
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
3: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```
OR
```
# tlp bat
TLP started in battery mode (manual).
# tlp-stat --rfkill
--- TLP 1.7.0 --------------------------------------------

+++ Wireless
bluetooth = on
nfc       = on
wifi      = on
wwan      = none (no device)

hci0(btusb)                   : bluetooth, connected
wlp0s20f3(iwlwifi)            : wifi, connected, power management = on

# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: nfc0: NFC
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
3: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```

OS: Fedora 42
HW: Lenovo Thinkpad X1 Gen 9
tlp-stat: https://gist.github.com/Coacher/f0810e682a55ff14a76a7fbc641df345

Coacher, 2025-06-18:
Yes, it does:
```
# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: nfc0: NFC
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
3: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no

# rfkill block nfc

# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: nfc0: NFC
        Soft blocked: yes
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
3: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```

Coacher, 2025-06-18:
Sorry, seems to be incorrect testing on my side. The problem persists in 1.8.0.

linrunner, 2025-06-19:
So `rfkill block nfc` in the running system works. What about TLP's command

    nfc disable
    nfc enable

?

Coacher, 2025-06-20:
```
[root@lenovo-x1: ~]# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: yes
        Hard blocked: no
1: nfc0: NFC
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
[root@lenovo-x1: ~]# nfc disable
nfc       = on
[root@lenovo-x1: ~]# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: yes
        Hard blocked: no
1: nfc0: NFC
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
[root@lenovo-x1: ~]# nfc enable
nfc       = on
[root@lenovo-x1: ~]# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: yes
        Hard blocked: no
1: nfc0: NFC
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
```

Coacher, 2025-06-20:
```
[root@lenovo-x1: ~]# tlp-stat -T
--- TLP 1.8.0 --------------------------------------------

июн 20 12:48:02 lenovo-x1 tlp[1995]: Applying power save settings...done.
июн 20 12:48:02 lenovo-x1 tlp[1995]: Setting battery charge thresholds...done.
июн 20 12:48:02 lenovo-x1 tlp[1995]: Disabling radios: nfc.
[root@lenovo-x1: ~]# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: nfc0: NFC
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
3: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
# Enabled TLP_DEBUG="rf run"
[root@lenovo-x1: ~]# nfc disable
nfc       = on
[root@lenovo-x1: ~]# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
        Soft blocked: no
        Hard blocked: no
1: nfc0: NFC
        Soft blocked: no
        Hard blocked: no
2: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
3: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
[root@lenovo-x1: ~]# tlp-stat -T
--- TLP 1.8.0 --------------------------------------------

июн 20 12:48:02 lenovo-x1 tlp[1995]: Applying power save settings...done.
июн 20 12:48:02 lenovo-x1 tlp[1995]: Setting battery charge thresholds...done.
июн 20 12:48:02 lenovo-x1 tlp[1995]: Disabling radios: nfc.
июн 20 12:54:41 lenovo-x1 tlp[4991]: device_state(nfc)
июн 20 12:54:41 lenovo-x1 tlp[4991]: get_devc(nfc) = /sys/class/rfkill/rfkill1/state
июн 20 12:54:41 lenovo-x1 tlp[4991]: get_devs(nfc) = 1
```

Coacher, 2025-06-20:
The output is rather lengthy, here's the gist:
https://gist.github.com/Coacher/bee3f9aa03f1a92e9af107cc075f8384

linrunner, 2025-06-20:
Oh my, what nonsense am I writing here :-(. Let's get it right:

     nfc off
     tlp-stat -T

Coacher, 2025-06-20:
```
--- TLP 1.8.0 --------------------------------------------

июн 20 15:02:03 lenovo-x1 tlp[2750]: parse_args4config: tlp auto --
июн 20 15:02:03 lenovo-x1 tlp[2750]: +++ auto (1.8.0) ++++++++++++++++++++++++++++++++++++++++
июн 20 15:02:03 lenovo-x1 tlp[2750]: SHELL=/usr/bin/bash; umask=0022
июн 20 15:02:03 lenovo-x1 tlp[2750]: power_source=ac
июн 20 15:02:03 lenovo-x1 tlp[2750]: manual_mode=n
июн 20 15:02:03 lenovo-x1 tlp[2750]: power_mode=ac
июн 20 15:02:03 lenovo-x1 tlp[2750]: set_radio_device_states(0): enable= disable=
июн 20 15:02:03 lenovo-x1 tlp[2750]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpbcYhOp -> /run/tlp/run.conf
июн 20 15:02:08 lenovo-x1 tlp[3199]: parse_args4config: tlp init start --
июн 20 15:02:08 lenovo-x1 tlp[3199]: +++ init start (1.8.0) ++++++++++++++++++++++++++++++++++++++++
июн 20 15:02:08 lenovo-x1 tlp[3199]: SHELL=/usr/bin/bash; umask=0022
июн 20 15:02:08 lenovo-x1 tlp[3199]: power_source=ac
июн 20 15:02:08 lenovo-x1 tlp[3199]: manual_mode=n
июн 20 15:02:08 lenovo-x1 tlp[3199]: power_mode=ac
июн 20 15:02:08 lenovo-x1 tlp[3199]: Applying power save settings...done.
июн 20 15:02:08 lenovo-x1 tlp[3199]: Setting battery charge thresholds...done.
июн 20 15:02:08 lenovo-x1 tlp[3199]: set_radio_device_states(start): enable= disable=nfc
июн 20 15:02:08 lenovo-x1 tlp[3199]: device_switch(nfc, off, , )
июн 20 15:02:08 lenovo-x1 tlp[3199]: get_devc(nfc) = /sys/class/rfkill/rfkill1/state
июн 20 15:02:08 lenovo-x1 tlp[3199]: get_devs(nfc) = 1
июн 20 15:02:08 lenovo-x1 tlp[3199]: device_switch(nfc, off).rfkill
июн 20 15:02:08 lenovo-x1 tlp[3199]: get_devs(nfc) = 0
июн 20 15:02:08 lenovo-x1 tlp[3199]: device_switch(nfc, off).ok: rc=0
июн 20 15:02:08 lenovo-x1 tlp[3199]: Disabling radios: nfc.
июн 20 15:02:08 lenovo-x1 tlp[3199]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpVd6XCk -> /run/tlp/run.conf
июн 20 15:02:31 lenovo-x1 tlp[4434]: parse_args4config: tlp-stat -T --
```
After reboot NFC was disabled. This is positive change in 1.8.0 compared to 1.7.0.

Unfortunately, unblocking NFC and then running tlp start/tlp bat or unplugging AC doesn't disable NFC.

Coacher, 2025-06-20:
```
--- TLP 1.8.0 --------------------------------------------

июн 20 15:39:57 lenovo-x1 tlp[1841]: Applying power save settings...done.
июн 20 15:39:57 lenovo-x1 tlp[1841]: Setting battery charge thresholds...done.
июн 20 15:39:57 lenovo-x1 tlp[1841]: Disabling radios: nfc.
июн 20 15:41:44 lenovo-x1 tlp[3494]: parse_args4config: nfc on --
июн 20 15:41:44 lenovo-x1 tlp[3494]: device_switch(nfc, on, , )
июн 20 15:41:44 lenovo-x1 tlp[3494]: get_devc(nfc) = /sys/class/rfkill/rfkill1/state
июн 20 15:41:44 lenovo-x1 tlp[3494]: get_devs(nfc) = 0
июн 20 15:41:44 lenovo-x1 tlp[3494]: device_switch(nfc, on).rfkill
июн 20 15:41:44 lenovo-x1 tlp[3494]: get_devs(nfc) = 1
июн 20 15:41:44 lenovo-x1 tlp[3494]: device_switch(nfc, on).ok: rc=0
июн 20 15:41:50 lenovo-x1 tlp[3546]: parse_args4config: tlp start --
июн 20 15:41:50 lenovo-x1 tlp[3546]: +++ start (1.8.0) ++++++++++++++++++++++++++++++++++++++++
июн 20 15:41:50 lenovo-x1 tlp[3546]: SHELL=/usr/bin/bash; umask=0022
июн 20 15:41:50 lenovo-x1 tlp[3546]: power_source=bat
июн 20 15:41:50 lenovo-x1 tlp[3546]: manual_mode=n
июн 20 15:41:50 lenovo-x1 tlp[3546]: power_mode=bat
июн 20 15:41:51 lenovo-x1 tlp[3546]: set_radio_device_states(1): enable= disable=bluetooth
июн 20 15:41:51 lenovo-x1 tlp[3546]: device_switch(bluetooth, off, , )
июн 20 15:41:51 lenovo-x1 tlp[3546]: get_devc(bluetooth) = /sys/class/rfkill/rfkill0/state
июн 20 15:41:51 lenovo-x1 tlp[3546]: get_devs(bluetooth) = 1
июн 20 15:41:51 lenovo-x1 tlp[3546]: device_switch(bluetooth, off).rfkill
июн 20 15:41:55 lenovo-x1 tlp[3546]: get_devs(bluetooth) = 0
июн 20 15:41:55 lenovo-x1 tlp[3546]: device_switch(bluetooth, off).ok: rc=0
июн 20 15:41:55 lenovo-x1 tlp[3546]: save_runconf.ok: /run/tlp/tlp-run.conf_tmp9kx1W1 -> /run/tlp/run.conf
июн 20 15:42:00 lenovo-x1 tlp[3907]: parse_args4config: tlp-stat -T --
```

As far as I can understand, tlp sees that bluetooth could be disabled, but ignores nfc.

linrunner, 2025-06-20:
OK, it's caused by TLP. Would have worked with _nfc_ in DEVICES_TO_DISABLE_ON_BAT, but unfortunately not in DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE.

Please apply this small change in **/usr/share/tlp/func.d/30-tlp-func-rf-sw**: https://github.com/linrunner/TLP/commit/55af5ce3d52c95a1347003d6359b8f182d0bed59

and re-test.

linrunner, 2025-06-25:
@Coacher Hi, have you been able to test yet? I still need your test result before I can merge the fix.

linrunner, 2025-06-25:
Great. Thanks for your patience in troubleshooting. The fix will be released with 1.9.0 (no ETA yet).

I do not need logs.

linrunner, 2025-11-05:
TLP 1.9.0 Beta is out. Please help with testing!
https://github.com/linrunner/TLP/issues/832