linrunner/TLP #809: [F42] Fedora tlp package forcibly unmasks services during upgrade

Coacher, 2025-06-18:
Hello.

Steps to reproduce:
1. Install tlp-1.7.0 on Fedora 42
2. Mask systemd-rfkill.service
3. Install tlp-1.8.0 from https://repo.linrunner.de

Observed result:
systemd-rfkill.service is unmasked

Expected result:
systemd-rfkill.service stays masked

Additional info:
Due to this code in tlp.spec:
```
%post
%systemd_post tlp.service
if [ $1 -eq 2 ] ; then
    systemctl unmask systemd-rfkill.service
    systemctl unmask power-profiles-daemon.service
fi
```
According to the above code and https://docs.fedoraproject.org/en-US/packaging-guidelines/Scriptlets/, systemd-rfkill.service will be unmasked on every package update. Looks like a bug. Why is this needed?

This also contradicts docs, which suggest to mask systemd-rfkill service and socket: https://linrunner.de/tlp/installation/fedora.html#service-units

This also creates poor UX, as tlp package forcibly alters user's choices without consent.

linrunner, 2025-06-18:
Thanks for your report.

Packages in the TLP repo for Fedora are not official Fedora packages and therefore do not have to adhere strictly to Fedora's rules.

The volunteer (not the same as the Fedora maintainer) who kindly provides these packages has included a feature here that is common to Debian, for example, and I'm not going to stop him. I am grateful to him for making up for Fedora's release deficit.

This specialty will therefore not be included in the official Fedora package.

TLP's docs are for the official package of course.

TL;DR: the packages are provided “as is”.

closed 2025-06-18 by linrunner

Coacher, 2025-06-18:
I appreciate the volunteer work. Can we at least discuss the possibility of the change?