linrunner/TLP #810: [Q] Can tlp and intel-lpmd run at the same time?

arturgawlik, 2025-06-23:
I can found in documentation that `tlp` e.g. is conflicting with `power-profiles-daemon`, but what about https://github.com/intel/intel-lpmd. Is it fine to run both `tlp` and `intel-lpmd` at the same time?

linrunner, 2025-06-30:
Hi, I have not yet dealt with **intel-lpmd**. Could you please just give it a try, I don't have the right hardware.

From a first look I don't see any overlap with what TLP does, so no obvious conflict. However, I would have expected this functionality to become part of the CPU scheduler in the kernel.

I would also be interested to know what actual difference intel-lpmd makes to power consumption. It should be most noticeable at low CPu load and with idle CPU.

Could you compare measurements in W with and without **intel-lpmd**? Use powertop when the desktop is idle and no applications are running (wait 5 minutes before reading)? Thanks in advance.

linrunner, 2025-08-29:
Added to the docs.