linrunner/TLP #812: [ThinkPad T14s Gen1 AMD] Unable to discharge/recalibrate the battery

antroids, 2025-06-30:
`$ sudo tlp recalibrate BAT0 
Setting temporary charge thresholds for battery BAT0:
  start =  96 (no change)
  stop  = 100 (no change)
Error: target level (0%) is too high compared to the actual charge level (100%) of battery BAT0.
`
`$ sudo tlp discharge BAT0 
Error: target level (0%) is too high compared to the actual charge level (100%) of battery BAT0.
`
`$ sudo tlp discharge BAT0 1
Error: target level (1%) is too high compared to the actual charge level (100%) of battery BAT0.
`
`$ sudo tlp recalibrate BAT0 
Setting temporary charge thresholds for battery BAT0:
  start =  96 (no change)
  stop  = 100 (no change)
Error: target level (0%) is too high compared to the actual charge level (99%) of battery BAT0.`

```
$ sudo tlp-stat 
--- TLP 1.8.0 --------------------------------------------

+++ Configured Settings:
defaults.conf L0006: TLP_ENABLE="1"
defaults.conf L0007: TLP_WARN_LEVEL="3"
defaults.conf L0008: TLP_MSG_COLORS="91 93 1 92"
defaults.conf L0009: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0010: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0011: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0012: MAX_LOST_WORK_SECS_ON_AC="15"
/etc/tlp.conf L0092: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0014: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
/etc/tlp.conf L0150: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0016: NMI_WATCHDOG="0"
defaults.conf L0017: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0018: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0019: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0020: DISK_APM_CLASS_DENYLIST="usb ieee1394"
defaults.conf L0021: DISK_IOSCHED="keep keep"
defaults.conf L0022: SATA_LINKPWR_ON_AC="med_power_with_dipm"
defaults.conf L0023: SATA_LINKPWR_ON_BAT="med_power_with_dipm"
defaults.conf L0024: AHCI_RUNTIME_PM_ON_AC="on"
defaults.conf L0025: AHCI_RUNTIME_PM_ON_BAT="auto"
defaults.conf L0026: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0027: PCIE_ASPM_ON_AC="default"
defaults.conf L0028: PCIE_ASPM_ON_BAT="default"
defaults.conf L0029: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0030: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0031: RADEON_POWER_PROFILE_ON_AC="default" #! Parameter is deprecated: feature will be removed with the next release
defaults.conf L0032: RADEON_POWER_PROFILE_ON_BAT="default" #! Parameter is deprecated: feature will be removed with the next release
defaults.conf L0033: AMDGPU_ABM_LEVEL_ON_AC="0"
defaults.conf L0034: AMDGPU_ABM_LEVEL_ON_BAT="1"
defaults.conf L0035: WIFI_PWR_ON_AC="off"
defaults.conf L0036: WIFI_PWR_ON_BAT="on"
defaults.conf L0037: WOL_DISABLE="Y"
defaults.conf L0038: SOUND_POWER_SAVE_ON_AC="1"
defaults.conf L0039: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0040: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0041: BAY_POWEROFF_ON_AC="0"
defaults.conf L0042: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0043: BAY_DEVICE="sr0"
defaults.conf L0044: RUNTIME_PM_ON_AC="on"
defaults.conf L0045: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0046: RUNTIME_PM_DRIVER_DENYLIST="mei_me nouveau radeon xhci_hcd"
defaults.conf L0047: USB_AUTOSUSPEND="1"
defaults.conf L0048: USB_EXCLUDE_AUDIO="1"
defaults.conf L0049: USB_EXCLUDE_BTUSB="0"
defaults.conf L0050: USB_EXCLUDE_PHONE="0"
defaults.conf L0051: USB_EXCLUDE_PRINTER="1"
defaults.conf L0052: USB_EXCLUDE_WWAN="0"
[... 178 lines cut ...]
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 02DL013
/sys/class/power_supply/BAT0/cycle_count                    =      3
/sys/class/power_supply/BAT0/charge_full_design             =  40780 [mWh] *
/sys/class/power_supply/BAT0/charge_full                    =  40780 [mWh] *
/sys/class/power_supply/BAT0/charge_now                     =  40780 [mWh] *
/sys/class/power_supply/BAT0/current_now                    =  25510 [mW] *
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =  100.0 [%]
Capacity                                                    =  100.0 [%]

*) Converted coreboot charge readings may differ.
```

linrunner, 2025-06-30:
Hi. This is _not_ a coreboot BIOS, or is it?

The occurrence of `/sys/class/power_supply/BAT?/charge_*` is an interesting singularity. 

Generally on ThinkPads there is only `/sys/class/power_supply/BAT?/energy_*`. TLP does not yet consider `charge_*` for checking the discharge target.

linrunner, 2025-07-02:
Well. I need to find a way to include this exception without bloating the code too much and I also need a new way to detect coreboot other than via the _charge_*_ attributes.  This will take quite a while. Please stay tuned.

linrunner, 2025-07-03:
Thanks to quick answers from the community, the coreboot detection could be adapted swiftly and the thinkpad plugin change was easier than expected.

Please replace **/usr/share/tlp/bat.d/05-thinkpad** with 
https://github.com/linrunner/TLP/blob/issue/812/bat.d/05-thinkpad
and test again.

**If it works now, please post the content of the progress screen.**

antroids, 2025-07-03:
```
$ sudo tlp-stat -s -b
--- TLP 1.8.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad T14s Gen 1 20UH001JRT
BIOS           = R1CET82W(1.51 )
EC Firmware    = 1.51
OS Release     = KDE neon User Edition
Kernel         = 6.11.0-28-generic #28~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri May 23 10:31:01 UTC 2 x86_64
/proc/cmdline  = initrd=\52bbcd908b354185ace5236e990e04d8\6.11.0-28-generic\initrd.img-6.11.0-28-generic root=UUID=1d2c1335-b1cb-4b51-b6b7-9fb62df7e174 ro quiet cryptdevice=UUID=688a1944-0341-472f-82e6-d3d503f60c67:luks-688a1944-0341-472f-82e6-d3d503f60c67 root=/dev/mapper/luks-688a1944-0341-472f-82e6-d3d503f60c67 splash resume=/dev/mapper/luks-7395464a-300d-4738-90e6-60f7bc105f06 vt.handoff=7 systemd.machine_id=52bbcd908b354185ace5236e990e04d8
Init system    = systemd v255 (255.4-1ubuntu8.8)
Boot mode      = UEFI
Suspend mode   = s2idle [deep]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 20:11:13, 2 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, chargeonce, discharge, recalibrate
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 02DL013
/sys/class/power_supply/BAT0/cycle_count                    =      5
/sys/class/power_supply/BAT0/charge_full_design             =   4078 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   4078 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   2701 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   66.2 [%]
Capacity                                                    =  100.0 [%]
```

Recalibrate doesn't work, but the reason is different:
```
$ sudo tlp recalibrate BAT0
[sudo] password for ak: 
Setting temporary charge thresholds for battery BAT0:
  start =  96 (no change)
  stop  = 100 (no change)
Initiating discharge of battery BAT0 ...............
Error: discharge BAT0 malfunction -- check your hardware (battery, charger).
```
Discharging itself works well, but the values look incorrect:
```
Currently discharging battery BAT0 to 0% (0 mWh):
voltage            =  10796 [mV]
remaining charge   =   2708 [mWh]
remaining percent  =   66.4 [%]
remaining time     =    290 [min]
current             =    560 [mW]
status             = Discharging
force-discharge    = 1
Press Ctrl+C to cancel.
```

[image]

linrunner, 2025-07-04:
@antroids I fixed the progress output. Please test again with:
https://github.com/linrunner/TLP/blob/issue/812/bat.d/05-thinkpad

Please continue testing with `tlp discharge` and also show your command and the initialization sequence.

The number of dots indicates how many seconds the EC needs to activate force-discharge. The time fluctuates significantly on my ThinkPads and it can be tight at times. Currently TLP limits it to 15 seconds. 

You may increase the value in [Line #950](https://github.com/linrunner/TLP/blob/issue/812/bat.d/05-thinkpad#L950)

antroids, 2025-07-07:
Now the discharge information looks much better:

`sudo tlp discharge BAT0`

```
Currently discharging battery BAT0 to 0% (0 mWh):
voltage            =  10539 [mV]
remaining energy   =  51910 [mWh]
remaining percent  =   99.7 [%]
remaining time     =    489 [min]
power              =   6365 [mW]
status             = Discharging
force-discharge    = 1
Press Ctrl+C to cancel.
```
The discharging is starting on second command run, in about 25 seconds after first discharge command. 
I broke something by these multiple discharge commands and now the battery shows 0% charge periodically.

```
$ sudo tlp-stat -s -b
--- TLP 1.8.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad T14s Gen 1 20UH001JRT
BIOS           = R1CET82W(1.51 )
EC Firmware    = 1.51
OS Release     = KDE neon User Edition
Kernel         = 6.11.0-28-generic #28~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri May 23 10:31:01 UTC 2 x86_64
/proc/cmdline  = initrd=\52bbcd908b354185ace5236e990e04d8\6.11.0-28-generic\initrd.img-6.11.0-28-generic root=UUID=1d2c1335-b1cb-4b51-b6b7-9fb62df7e174 ro quiet cryptdevice=UUID=688a1944-0341-472f-82e6-d3d503f60c67:luks-688a1944-0341-472f-82e6-d3d503f60c67 root=/dev/mapper/luks-688a1944-0341-472f-82e6-d3d503f60c67 splash resume=/dev/mapper/luks-7395464a-300d-4738-90e6-60f7bc105f06 vt.handoff=7 systemd.machine_id=52bbcd908b354185ace5236e990e04d8
Init system    = systemd v255 (255.4-1ubuntu8.8)
Boot mode      = UEFI
Suspend mode   = s2idle [deep]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 20:54:14, 14 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, chargeonce, discharge, recalibrate
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 02DL013
/sys/class/power_supply/BAT0/cycle_count                    =      6
/sys/class/power_supply/BAT0/energy_full_design             =  53010 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  52070 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  51520 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   5891 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =   98.9 [%]
Capacity                                                    =   98.2 [%]
```

```
$ sudo tlp-stat -s -b
--- TLP 1.8.0 --------------------------------------------

+++ System Info
System         = LENOVO ThinkPad T14s Gen 1 20UH001JRT
BIOS           = R1CET82W(1.51 )
EC Firmware    = 1.51
OS Release     = KDE neon User Edition
Kernel         = 6.11.0-28-generic #28~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri May 23 10:31:01 UTC 2 x86_64
/proc/cmdline  = initrd=\52bbcd908b354185ace5236e990e04d8\6.11.0-28-generic\initrd.img-6.11.0-28-generic root=UUID=1d2c1335-b1cb-4b51-b6b7-9fb62df7e174 ro quiet cryptdevice=UUID=688a1944-0341-472f-82e6-d3d503f60c67:luks-688a1944-0341-472f-82e6-d3d503f60c67 root=/dev/mapper/luks-688a1944-0341-472f-82e6-d3d503f60c67 splash resume=/dev/mapper/luks-7395464a-300d-4738-90e6-60f7bc105f06 vt.handoff=7 systemd.machine_id=52bbcd908b354185ace5236e990e04d8
Init system    = systemd v255 (255.4-1ubuntu8.8)
Boot mode      = UEFI
Suspend mode   = s2idle [deep]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 20:54:14, 15 sec(s) ago
Mode           = AC
Power source   = AC

+++ Battery Care
Plugin: thinkpad
Supported features: charge thresholds, chargeonce, discharge, recalibrate
Driver usage:
* natacpi (thinkpad_acpi) = active (charge thresholds, recalibration)
Parameter value ranges:
* START_CHARGE_THRESH_BAT0/1:  0(off)..96(default)..99
* STOP_CHARGE_THRESH_BAT0/1:   1..100(default)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 02DL013
/sys/class/power_supply/BAT0/cycle_count                    =      6
/sys/class/power_supply/BAT0/energy_full_design             =  53010 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  52070 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =      0 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   5890 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

/sys/class/power_supply/BAT0/charge_control_start_threshold =     96 [%]
/sys/class/power_supply/BAT0/charge_control_end_threshold   =    100 [%]
/sys/class/power_supply/BAT0/charge_behaviour               = [auto] inhibit-charge force-discharge

Charge                                                      =    0.0 [%]
Capacity                                                    =   98.2 [%]
```
And current Charge indicator is disappearing from KDE Energy Information.

linrunner, 2025-07-08:
> It's totally fine after reset. 

Glad to hear that.

> I have third party aftermarket battery, but not sure that this could be a reason. 

Good luck with that, they tend to cause trouble.

> I think this first generation may have some issues with battery controller software/hardware.

Nope, "Gen X" was just a name scheme change for marketing purposes. There were predecessors with AMD like the T495(s) and the A series.

However, the EC seems to lose its mind temporarily from time to time:
https://linrunner.de/tlp/faq/battery.html#various-thinkpad-battery-malfunctions

closed 2025-07-08 by linrunner