linrunner/TLP #813: TLP disables Wifi if started when laptop is docked

BladeMF, 2025-07-03:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**
When running `sudo tlp start` and the laptop is connected to a docking station, the WiFi is disabled. If I then disconnect and reconnect the docking station, the WiFi remains enabled.

**Expected behavior**
WiFi to remain enabled.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
> Only when connected to a docking station with charging
3. Actions to reproduce the behaviour
> 1) Connect the laptop to a docking station
> 2) Run `sudo tlp start`
5. Shell commands entered and their output *including error message(s)*
> `sudo tlp start`. No output.
6. **Full output of `tlp-stat` via https://gist.github.com/ for *all*   matching cases of 1** (not as file attachment, no screenshots)
> https://gist.github.com/BladeMF/53926d395110482f0e2b5c86f4d4a476
**Additional context**

_No additional context._

linrunner, 2025-07-04:
Hi. The output you provide does not match the problem situation you describe:

> +++ Wireless
>
> **wifi      = on**
> 
> wlp2s0f0(iwlwifi)             : wifi, **connected**, power management = off

Wi-Fi is obviously on and connected.

> Last run       = 10:28:32, 513 sec(s) ago

Boot was 8 minutes ago. I expect tlp-stat output at the time the problem occurs (or shortly after).

Is it a USB-C / Thunderbolt dock? 

Have you tried deactivating TLP completely i.e. `TLP_ENABLE=0`  - does the issue then disappear?

linrunner, 2025-07-07:
> I've updated the gist the the output. It's just after boot and the Wifi was _definitely_ off. 

In fact, it is switched on on the rfkill side, but not connected:

> +++ Wireless
> ...
> wifi      = on
> ...
> wlp2s0f0(iwlwifi)             : wifi, not connected, power management = off

I suspect an unforeseen side effect - possibly some power saving - on NetworkManager, because with your current configuration TLP won't touch Wifi at all.

> It appeared off in Ubuntu and clicking on it did not turn it on.

How exactly did you then reactivate/reconnect Wifi?

> Also, is there any point in trying with ENABLE=false if the problem is demonstrated when I run `sudo tlp start` at any point when the laptop is docked?

I had misread your initial post and assumed that the problem occurs at system startup. Happens you now confirmed that in addition to your demo scenario with `tlp start`.  

Since I find your symptom quite puzzling, I would like to continue comparing system startup with and without TLP. 

I now ask you to perform a total of three test cases with output.

**Prequisites**
First activate the trace via configuration (remains the same for all cases):

    TLP_DEBUG="arg bat disk lock nm path pm ps rf run sysfs udev usb"

Laptop is connected to the dock during all tests.

**Test 1: Cold Boot with TLP**
Note: in principle, we already have it, but I also need the corresponding trace and NM status.
After boot show:

     nmcli device status
     sudo tlp-stat
     journalctl -p debug -t tlp -t NetworkManager -b 0

     
**Test 2: Cold Boot without TLP**
Change config to `TLP_ENABLE=0` shutdown and power on again. 
After boot show:

     nmcli device status
     sudo tlp-stat
     journalctl -p debug -t tlp -t NetworkManager -b 0

**Test 3: Your tlp start Demo Case**
Revert config to `TLP_ENABLE=1`.
Make Wifi connected.

     nmcli device status
     sudo tlp start
     nmcli device status
     sudo tlp-stat
     journalctl -p debug -t tlp -t NetworkManager -b 0

BladeMF, 2025-07-08:
Sure, I'll try and do them tomorrow.

> How exactly did you then reactivate/reconnect Wifi?
I unplug the dock, the WiFi gets enabled and starts connecting instantly and then plug it back in. It takes no more than a second. No plugging or unplugging from this point on affects WiFi.

linrunner, 2025-07-08:
Well, this means that Wifi becomes inoperable in the AC profile, while the BAT profile re-activates it. Interesting, it's often the other way around. Then I also need the outputs for this state:

**Test 4: undocked / on battery power**
Show:

     nmcli device status
     sudo tlp-stat

BladeMF, 2025-07-09:
There you go. Interesting note - when on login screen I _always_ have WiFi, whether TLP is enabled or not. 
_Note_: Test 3 is done on the same boot directly after Test 2 (Since the WiFi was on and connected).

[test1.txt](https://github.com/user-attachments/files/21134920/test1.txt)
[test1-journal.txt](https://github.com/user-attachments/files/21134919/test1-journal.txt)
[test2.txt](https://github.com/user-attachments/files/21134917/test2.txt)
[test2-journal.txt](https://github.com/user-attachments/files/21134921/test2-journal.txt)
[test3.txt](https://github.com/user-attachments/files/21134918/test3.txt)
[test3-journal.txt](https://github.com/user-attachments/files/21134922/test3-journal.txt)
[test4.txt](https://github.com/user-attachments/files/21134916/test4.txt)

linrunner, 2025-07-09:
The journal for test3 shows interesting details:

>  line 283:  юли 09 07:31:00 vkosev-u tlp[11526]: set_radio_device_states(0): enable= disable=

TLP does not touch WiFi - works as designed, you have not configured anything. But then, out of the blue:

> line 286: юли 09 07:31:00 vkosev-u NetworkManager[2243]: <info>  [1752035460.2561] manager: rfkill: Wi-Fi now disabled by radio killswitch

NetworkManager disables WiFi. Why??

Same thing is happening in test1 journal:

> line 282 юли 09 07:17:57 vkosev-u NetworkManager[2771]: <info>  [1752034677.9674] manager: rfkill: Wi-Fi now disabled by radio killswitch

It is also worth noting that despite `WIFI_PWR_ON_BAT="on"`, WiFi power save stays off in test 4 on battery:

> wlp2s0f0(iwlwifi)             : wifi, connected, power management = off

Could the driver be blocking it?

I have not the slightest idea what is happening on your machine. I'd try to deactivate Runtime PM completely:

    RUNTIME_PM_ON_AC=""
    RUNTIME_PM_ON_BAT=""

Then run test1 again and provide the outputs.

-----------

> Interesting note - when on login screen I _always_ have WiFi, whether TLP is enabled or not. 

This is an important hint. We have to ask: what happens during the graphical login? TLP definitely does nothing. Does your desktop environment have a power saving feature for WiFi?

-----------

BladeMF, 2025-07-09:
> Then run test1 again and provide the outputs.
OK

> This is an important hint. We have to ask: what happens during the graphical login? TLP definitely does nothing. Does your desktop environment have a power saving (or switching) feature for WiFi?

Isn't TLP starting on login?

BladeMF, 2025-07-09:
Well _that's_ interesting!

> Does your desktop environment have a power saving (or switching) feature for WiFi?

|Nothing I put. It's a stock Ubuntu. Nothing fancy.

linrunner, 2025-07-09:
Anyway. We continue with the config change and test1 outputs.

BladeMF, 2025-07-11:
There you go:

[test5.txt](https://github.com/user-attachments/files/21176943/test5.txt)
[test5-journal.txt](https://github.com/user-attachments/files/21176942/test5-journal.txt)

linrunner, 2025-07-11:
Thanks. This remains mysterious to me. 

Here TLP starts to apply its settings for the first time:

> юли 11 09:34:12 vkosev-u tlp[1418]: +++ auto (1.8.0) ++++++++++++++++++++++++++++++++++++++++

But in this moment there are no W-Fi interfaces present:

> юли 11 09:34:13 vkosev-u tlp[1418]: set_wifi_power_mode(0).no_ifaces

The actual Wi-Fi "black out" happens 14 secs later:

> юли 11 09:34:27 vkosev-u wpa_supplicant[2655]: rfkill: WLAN **hard blocked**
> юли 11 09:34:27 vkosev-u wpa_supplicant[2655]: rfkill: WLAN hard blocked
> юли 11 09:34:27 vkosev-u kernel: iwlwifi 0000:02:00.0: RF_KILL bit toggled to disable radio.
> юли 11 09:34:27 vkosev-u kernel: iwlwifi 0000:02:00.0: reporting RF_KILL (radio disabled)
> юли 11 09:34:27 vkosev-u NetworkManager[2648]: <info>  [1752215667.9292] manager: rfkill: Wi-Fi now disabled by radio killswitch

Please note that it is a *hard* block which cannot be achieved by software (rfkill block), but requires a switch or a button on the laptop. Maybe something in the kernel gets mixed up at this point.

You have described above how to reactivate Wi-Fi:

> I unplug the dock, the WiFi gets enabled and starts connecting instantly and then plug it back in. It takes no more than a second. No plugging or unplugging from this point on affects WiFi.

BladeMF, 2025-07-15:
There you go.
[test6.txt](https://github.com/user-attachments/files/21226484/test6.txt)

P.S. I've obfuscated the IPs and removed the links for the previous tests. HMU if you need them.

linrunner, 2025-07-15:
Here WiFi is reactivated on the hardware side:
```
юли 15 08:50:51 vkosev-u kernel: iwlwifi 0000:02:00.0: RF_KILL bit toggled to enable radio.
юли 15 08:50:51 vkosev-u kernel: iwlwifi 0000:02:00.0: reporting RF_KILL (radio enabled)
```

Could a BIOS setting be the cause? See the answer here (at the end):
https://askubuntu.com/questions/1149113/iwlwifi-auto-disabling-radio

BladeMF, 2025-07-22:
Yeah, sorry mate, been a bit busy. Gonna try right now and write back.

BladeMF, 2025-07-22:
Yep, that did it. The docking station _does_ have a WAN port, but it is unplugged. Is it possible that TLP somehow tricks the firmware into thinking it is plugged?

linrunner, 2025-07-22:
@BladeMF Great. I'm glad we could figure it out. I will add it to the FAQ.

> Is it possible that TLP somehow tricks the firmware into thinking it is plugged?

I guess so and consider it a firmware bug. The big question is: what setting is it? It must be one that has different values for _AC and _BAT.

If you really want to know and have a lot of time, you could try out step by step what happens when you set _AC = _BAT for individual settings. I would start with these two candidates:

```
/etc/tlp.d/blademf.conf L0013: PLATFORM_PROFILE_ON_AC="performance" # try "balanced" then "low-power"
/etc/tlp.d/blademf.conf L0014: PLATFORM_PROFILE_ON_BAT="low-power"

/etc/tlp.d/blademf.conf L0007: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance" # try "balance_power" or "power"
/etc/tlp.d/blademf.conf L0008: CPU_ENERGY_PERF_POLICY_ON_BAT="power"

```
Good luck. Please post if you find something out.

BladeMF, 2025-08-08:
Thanks. I keep having the intention of finding the root cause, but I have no realistic expectation of when will I have the time to do it.