linrunner/TLP #821: platform_profile fails to change automatically upon change in power source

bound-variable, 2025-08-02:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've provided all required `tlp-stat` outputs via [Gist](https://gist.github.com/) (see below)

**Describe the bug**

When my system is powered by AC and I run `tlp-stat -p` I see the platform_profile set at "performance". When I unplug the AC, platform_profile changes to "balanced". But when I plug the AC back in, platform_profile remains set at "balanced".

The other AC settings do change correctly according to the configuration.

**Expected behavior**

The setting platform_profile should change automatically upon change in power source and according to the configuration.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. The problem occurs upon change of power source from battery to AC.
2. Run tlp start. Check output of tlp-stat -p. Unplug AC. Check output of tlp-stat -p again. Plug in AC. Check output of tlp-stat -p again.
3. No error messages are displayed. 
5. Output of tlp-stat: https://gist.github.com/bound-variable/adfe70fe28e7872f5a6352be471d3212

linrunner, 2025-08-02:
@bound-variable You haven't also installed **tuned** and **tuned-ppd** (or **power-profiles-daemon**), have you? Both also change the profile. In that case, remove them please.

bound-variable, 2025-08-03:
Okay, I had **tuned** installed. I uninstalled it. Rebooted. But now whenever I change the power source (plug/unplug AC) then the **energy_performance_preference** is fixed on the AC setting (performace). It doesn't change to the BAT setting (power) when unplugging the AC.

linrunner, 2025-08-07:
There is no switch to the battery profile at all:

> +++ TLP Status
> **Mode           = AC**
> Power source   = battery

All settings remain on the AC values. 

I can only assume that, as so often in the past, SELinux has been messed up again and is blocking the application of the settings.

You may check for AVCs with 

     sudo ausearch -su 'tlp' -ts boot

linrunner, 2025-08-09:
So as already suspected, a regression in Fedora. Please open a Fedora bugreport against the package **selinux-policy**. Post the link here. Good luck!

bound-variable, 2025-08-11:
Don't know if I can do that because I don't intend on remaining with Fedora for long. I see I can just disable selinux and tlp works fine. Since I have no need for selinux, I think I'll conclude the matter here.